seemoo-lab/nexmon_csi #2: Need help setting up the csi parameters

TheFreakyHollow, 2019-10-01:
Hey guys,

I just started setting up the CSI tool on my RasPi 4B running kernel version 4.19 and had no errors installing it. It also seems to be working when I copy the parameter string you gave in the example and exchange the source mac address with my routers mac address to gather the connections csi-info (wich I cannot confirm, because the router is using a different channel/other routers around use the 2.4 GHz band). So the tcpdump doesn't dump anything since there is no traffic on channel 157.

I'd like to know a few things...

...where I can find the mentioned "broadcom format channel specification" so I can listen on different bands/channels or use different bandwidths?
Searching for the example string "0xe29b" or "broadcom format channel specification" didn't get me anywhere. Neither did reading the readme. I am assuming there is a way to convert this info into a sting (hexdec) as given above.

...do I have to be connected to the AccessPoint? Or is receiving just its beacon fine? I'd rather not connect to it because I might not be able to do so in the future when analyzing different streams from different routers.

... I am currently not using the "-b" parameter. Is the default setting that it simply dumps all frames?

Let me know if you need any further information :)

thanks a lot in advance and for creating the tool,
Benedikt

jlinktu, 2019-10-02:
Hi Benedikt,

that's a good point. You can get the hex value by setting the chanspec using nexutil and then reading out the current chanspec with nexutil e.g. for channel 157 with 80MHz bandwidth:
```
$ nexutil -k157/80 && nexutil -k
chanspec: 0xe29b, 157/80
```
For more information have a look at `$ nexutil --help`. If you want to know more about the conversion itself you can have a look at the nexutil source code.
We will introduce a feature for the makecsiparams tool that makes it easier to set the channel in the future.

No, you do not have to be connected to an AP, just make sure monitor mode is enabled as described in the readme usage section.

Correct, no _-b_ parameter means no byte filtering.

TheFreakyHollow, 2019-10-07:
Thank you for your fast answer and for pointing out that there is the `$ nexutil --help` option...
I can change the chanspec to Ch 157 with 80 MHz BW without any problems, but I can't seem to chance it to other channels. I am using this:

https://en.wikipedia.org/wiki/List_of_WLAN_channels#cite_note-ERC.Recommendation.70-03.Annex1-55

Wikipedia-List to see what channel ist using wich frequency (I'd like to go as high as possible i.e. ch159+).
What am I missing? Why is `$ nexutil -k142/20` or `$ nexutil -k142/40` not working?
I do not get an error message, but the chanpec doesn't change. For lower channels something like  `$ nexutil -k7` works (without giving any bandwith).

thanks again!
Benedikt

matthiasseemoo, 2019-10-07:
You might need to add your channel settings to the regulations.c file. By
default you cannot set completely arbitraty channels.

TheFreakyHollow, 2019-10-07:
I am assuming all I need to do is to add my hex value in here?

`#if (NEXMON_CHIP == CHIP_VER_BCM43455c0)
    0xe02a, // 36/80
    0xe29b, // 157/80
#endif
`
If so, the above problem occurs: is there an easy way to get the hex value for my wanted configuration?

jlinktu, 2019-10-07:
You can also use the macros defined in _nexmon/patches/include/channels.h_ e.g.:
`CH20MHZ_CHSPEC(142)`

TheFreakyHollow, 2019-10-07:
I did add the line you gave as an example. The regulations.c file now states:
```
#if (NEXMON_CHIP == CHIP_VER_BCM43455c0)
    0xe02a, // 36/80
    0xe29b, // 157/80
    CH20MHZ_CHSPEC(142),
#endif
```
I'm guessing I have to recompile the code? Just saving this + rebooting didn't do it.
thanks again!

jlinktu, 2019-10-07:
Yes, you have to recompile and reinstall the firmware, run `make && make install-firmware` again inside the _nexmon\_csi_ patch directory.

TheFreakyHollow, 2019-10-08:
Thank you for the instructions.
When I try to `$ sudo make && make install-firmware`
the following error (in german) pops up:
```
pi@raspberrypi:~/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi $ sudo make
Makefile:3: /firmwares/bcm43455c0/7_45_189/definitions.mk: Datei oder Verzeichnis nicht gefunden
make: *** Keine Regel, um „/firmwares/bcm43455c0/7_45_189/definitions.mk“ zu erstellen.  Schluss.

```
But finding gives me the following:
```
pi@raspberrypi:~/nexmon $ find -name definitions.mk
./firmwares/bcm4358/7_112_200_17_sta/definitions.mk
./firmwares/bcm4358/7_112_300_14_sta/definitions.mk
./firmwares/bcm4358/7_112_201_3_sta/definitions.mk
./firmwares/bcm43455/7_45_77_0_hw_23_8_2017/definitions.mk
./firmwares/bcm43455/7_45_77_0_hw/definitions.mk
./firmwares/bcm43455/7_120_7_1_sta_C0/definitions.mk
./firmwares/bcm43455/7_120_5_1_sta_C0/definitions.mk
./firmwares/bcm4361b0/13_38_55_1_sta/definitions.mk
./firmwares/bcm43430a1/7_45_41_46/definitions.mk
./firmwares/bcm43430a1/7_45_41_26/definitions.mk
./firmwares/bcm4339/6_37_34_43/definitions.mk
./firmwares/bcm43451b1/7_63_43_0/definitions.mk
./firmwares/bcm43455c0/7_45_154/definitions.mk

./firmwares/bcm43455c0/7_45_189/definitions.mk  ( <-- this should be it )

./firmwares/bcm43596a0/9.96.4_sta_c0/definitions.mk
./firmwares/bcm43596a0/9.75.155.45_sta_c0/definitions.mk
./firmwares/bcm4335b0/6.30.171.1_sta/definitions.mk
./firmwares/bcm43438/7_45_41_26/definitions.mk
./firmwares/bcm4330/5_90_100_41_sta/definitions.mk
./firmwares/bcm4356/7_35_101_5_sta/definitions.mk

```
Is there any fix to that or should I do a clean install?
Cheers!
Benedikt

jlinktu, 2019-10-08:
Remember to set the environment every time before doing any compile related work, e.g. by running `source ./setup_env.sh` inside the nexmon root directory. No need for _sudo_ at this point, _make_ will ask you for it when it is required. If you want to use _sudo_ anyways remember to preserve the environment with `sudo -E`.

TheFreakyHollow, 2019-10-08:
Thanks a lot for the great support and your quick replies!

It did ask me to `sudo make`, so preserving the environment did the job.
However, after rebooting and trying it again, I am getting a linker error, even if I reverse all the changes made to __regulations.c__ .
After setting up the environment as before, I do now get:
```
pi@raspberrypi:~/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi $ sudo -E make

          ###########   ###########   ##########    ##########
         ############  ############  ############  ############
         ##            ##            ##   ##   ##  ##        ##
         ##            ##            ##   ##   ##  ##        ##
         ###########   ####  ######  ##   ##   ##  ##    ######
          ###########  ####  #       ##   ##   ##  ##    #    #
                   ##  ##    ######  ##   ##   ##  ##    #    #
                   ##  ##    #       ##   ##   ##  ##    #    #
         ############  ##### ######  ##   ##   ##  ##### ######
         ###########    ###########  ##   ##   ##   ##########

            S E C U R E   M O B I L E   N E T W O R K I N G

                               presents:

              # ###   ###  #   # # ###  ###   ###  # ###
              ##   # #   #  # #  ##   ##   # #   # ##   #
              #    # #####   #   #    #    # #   # #    #
              #    # #      # #  #    #    # #   # #    #
              #    #  #### #   # #    #    #  ###  #    #

                The C-based Firmware Patching Framework

                           !!! WARNING !!!
    Our software may damage your hardware and may void your hardware’s
     warranty! You use our tools at your own risk and responsibility

  COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  PREPARING gen/nexmon.pre => gen/nexmon2.pre
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
make: *** [Makefile:144: gen/patch.elf] Fehler 1
```
Please help.

jlinktu, 2019-10-08:
What do the last few lines of _log/linker.err_ say?

TheFreakyHollow, 2019-10-08:
Im not shure what exactly you need, so i copied the error file.

[linker.txt](https://github.com/seemoo-lab/nexmon_csi/files/3702098/linker.txt)

jlinktu, 2019-10-08:
Try a `make clean` first. If that does not help check for any introduced code changes (e.g. with `git diff`) and eliminate faults.

TheFreakyHollow, 2019-10-08:
In some way I messed up the regulations.c file and redownloaded it. I can now add the channel macros and `make && make install-firmware!`

It now looks like this:
```
#if (NEXMON_CHIP == CHIP_VER_BCM43455c0)
    0xe02a, // 36/80
    0xe29b, // 157/80
    CH20MHZ_CHSPEC(149),
    CH40MHZ_CHSPEC(149, WL_CHANSPEC_CTL_SB_L),
    CH80MHZ_CHSPEC(149, WL_CHANSPEC_CTL_SB_L),
#endif
};
```
I still can't seem to change the bandwith. I collected some data from my P20 Pro on channel 149, but the .pcap data only consisted of 20 MHz bandwith, as the matlabplot I did afterwards shows.

(in hindsight this was obvious because the nexutil tool gave the same hexdec string for different paramters):
```
pi@raspberrypi:~ $ nexutil -k149/20 && nexutil -k
chanspec: 0xd095, 149
pi@raspberrypi:~ $ nexutil -k149/40 && nexutil -k
chanspec: 0xd095, 149
pi@raspberrypi:~ $ nexutil -k149/80 && nexutil -k
chanspec: 0xd095, 149

```
I just blindly copied the macros. I am assuming that the other argument the function is talking stands for the upper sideband? Should I have added something different?

jlinktu, 2019-10-08:
The second argument is the control sideband, ending \_L means lower and \_U means upper.

When you tried setting the 40 and 80Mhz channels the chip actually never changed the current channel, but did not complain as the requested chanspecs were found in that list.
However, the current state of the regulations patch for the bcm43455c0 might lack some features so that you can not use completely arbitrary chanspecs by only adding them to that list. So for now you'll have to test and go with the chanspecs that are working.

TheFreakyHollow, 2019-10-08:
Thank you guys for guiding me throgh this! If I stumble upon anything else I'll let you know.
For now I'll just see which channels work best for me. Have a great night!

closed 2019-10-08 by TheFreakyHollow