seemoo-lab/nexmon_csi #3: Filtering with CSI parameters (-b help)

nikolafelkaroski, 2019-10-09:
Thank you for this great project.

I need a bit of help with the CSI parameters.

I tried to capture packets without using the -b argument and it works. But for an example I would like to capture only certain UDP or ICMP packets coming from the specified MAC. The help says that -b is the starting byte of the frame, but trying to input the first byte of the wanted packets that I see in Wireshark (on PC), results in 0 packets captured (using tcpdump on the PI3B+), so my question is exactly which byte does -b filter by?

example UDP packet:
**mac1**: aa:bb:aa:bb:aa:bb
**mac2**: 11:22:33:44:55:66
**UDP header**:c28b00a2000f0c83
**UDP data**: ababababababab

From wireshark:
------------------------------------------------------------------------
0000   aa bb aa bb aa bb 11 22 33 44 55 66 08 00 45 00
0010   00 23 ae ab 40 00 40 11 0a 03 c0 a8 00 65 c0 a8
0020   00 66 c2 8b 00 a2 00 0f 0c 83 ab ab ab ab ab ab
0030   ab

jlinktu, 2019-10-12:
Hi,
the argument to -b is compared to the starting byte of incoming 802.11 frames for filtering, thus defining version, type, and subtype fields of the frame control field. Filtering for lower protocol layers is currently not implemented.

nikolafelkaroski, 2019-10-13:
Thanks a lot.

closed 2019-10-13 by nikolafelkaroski