seemoo-lab/nexmon_csi #6: Will this tool support any other 11ac router beside AC86U?

quantumhub, 2019-10-31:
Thanks for this great work, which brings us 80MHz 4x4MIMO for the first time!
Asus AC5300 and AC88U are using the same chip as ac86u. It is also bcm4366. Will this tool support this router. AC5300 seems to able to collect 160MHz 4x4 via 2x 4366 chips.
And some earlier 11ac wave 1 MIMO 3x3 routers are cheap now. Will this project unlock the CSI readout capability of these routers.

matthiasseemoo, 2019-11-01:
The firmware patch itself generally supports all bcm4366c chips. For
getting 160 MHz bandwidth, one of those chips should be sufficient,
however, we did not yet understand how to extract CSI for it. In the end,
it seems like two 80 MHz transceivers in one chip are used in parallel.
Generally, any Broadcom chip should work to extract CSI but we do not have
the time to build the patch for every Broadcom chip. If you have a specific
demand, feel free to contact us via email to discuss it.

quantumhub, 2019-11-01:
Dear Matthias,

Thanks for your reply!
Just notice that recent AC86U model is using BCM4366E (a upgration from BCM4366C0?):

https://wikidevi.com/wiki/ASUS_RT-AC86U

Is it compatible with nexmon's existing patch?
Is there any instruction to create a new patch for a chip not yet covered by the project currently, so that people can contribute to it? Thank you!

bachejames, 2020-04-29:
> The firmware patch itself generally supports all bcm4366c chips. For getting 160 MHz bandwidth, one of those chips should be sufficient, however, we did not yet understand how to extract CSI for it. In the end, it seems like two 80 MHz transceivers in one chip are used in parallel. Generally, any Broadcom chip should work to extract CSI but we do not have the time to build the patch for every Broadcom chip. If you have a specific demand, feel free to contact us via email to discuss it. quantumhub [notifications@github.com](mailto:notifications@github.com) schrieb am Fr., 1. Nov. 2019, 00:01:

Would be really keen to know the answer to this one. @quantumhub did you ever try it out please? Looking at [this](https://en.wikipedia.org/wiki/List_of_Asus_routers) list of Asus routers, possibly `RT-AC88U`, `RT-AC3100`, `RT-AC5300` and `GT-AC5300` may all work out-of-the-box if any variant of the 4366 chip is supported?

Many thanks!

bachejames, 2020-04-30:
@jlinktu - wondered if you may be able to assist here please?

Very many thanks!

jlinktu, 2020-05-05:
Hi @bachejames,

the firmware itself should work for all devices as long as they have the same wifi chip connected via the same bus type, however, for the asus routers you have to encapsulate it inside a dhd kernel module that might not be interchangeable between the devices. But I can not confirm this as I do not have any of the devices. I you have one, you might want to test it and report it here :)

mikeWangJC, 2020-06-03:
keep my eyes on this topic

jlinktu, 2020-06-04:
support confirmed for Asus GT-AC5300 (by maxedbeech)

mzakharo, 2020-06-05:
https://wikidevi.wi-cat.ru/List_of_MU-MIMO_supported_devices  lists quite a few entries for ```BCM4366E```

hanabhp, 2020-07-09:
Thanks for your excellent work! I'm still wondering if nexmon's patch is compatible with AC86U model using BCM4366E?

jlinktu, 2020-07-09:
It is.

ML34, 2025-05-02:
Hi! I've been searching for an ASUS RT-AC86U with the BCM4366C0 chipset but haven’t had any luck — most newer units (same model name) seem to have a different chip (e.g., boardtype 0x0832). Are there any other routers currently supported by Nexmon CSI that I could consider instead? Thanks in advance!