seemoo-lab/nexmon_csi #10: CSI data for half the BW is missing

nikolafelkaroski, 2019-11-06:
Hello,

I am using a 5GHz router and a RP3B+ for CSI collection. My router is set to channel 40 with 80MHz bandwidth and I am trying to collect ICMP packets (-b 0x88) by pinging the router.
When I analyze the pcap files in Matlab, half of the subcarriers are zero (or noise close to zero). If I switch to capturing beacon frames (-b 0x80), only a quarter of the subcarriers show non-zero CSI data. Is this a router issue or is there a way to use packets which will generate CSI for all subcarriers?

Thank you for your time.

matthiasseemoo, 2019-11-06:
Even if you are on a 40 MHz wide channel, it is possible to send packets
with only 20 MHz bandwidth, normally the control frames. To ensure that you
transmit packets with 40 MHz bandwidth, inject them with nexmon and
explicitly select the bandwidth you want to use.

nikolafelkaroski, 2019-11-06:
Thank you for the help.

closed 2019-11-06 by nikolafelkaroski

zeroby0, 2020-02-13:
@matthiasseemoo can you please show me a quick pointer / link on how to inject packets with Nexmon with an explicit bandwidth set?

matthiasseemoo, 2020-02-18:
In the rates.h file you can see how to define rates:
https://github.com/seemoo-lab/nexmon/blob/master/patches/include/rates.h <https://github.com/seemoo-lab/nexmon/blob/master/patches/include/rates.h>

In the channels.h you can see how to define a channel so that it supports your bandwidth requirements:
https://github.com/seemoo-lab/nexmon/blob/master/patches/include/channels.h <https://github.com/seemoo-lab/nexmon/blob/master/patches/include/channels.h>

In case, your required chanspec is not supported by default, you need to add it in the regulations.c file:
https://github.com/seemoo-lab/wisec2017_nexmon_jammer/blob/9eaa0e6f68bd2cd92245ae4a845687ba14668c0f/src/regulations.c <https://github.com/seemoo-lab/wisec2017_nexmon_jammer/blob/9eaa0e6f68bd2cd92245ae4a845687ba14668c0f/src/regulations.c>

You can call the sendframe function to send frames and define the rates:
https://github.com/seemoo-lab/wisec2017_nexmon_jammer/blob/9eaa0e6f68bd2cd92245ae4a845687ba14668c0f/src/sendframe.c#L57 <https://github.com/seemoo-lab/wisec2017_nexmon_jammer/blob/9eaa0e6f68bd2cd92245ae4a845687ba14668c0f/src/sendframe.c#L57>

mikeWangJC, 2020-05-05:
@matthiasseemoo 
hi, buddy, i'm trying to inject packet with the  select the bandwidth and frequency.
I use 'tcpdump', but no packet is received. 
do you hav any suggestion? thank yuo so much.