seemoo-lab/nexmon_csi #11: Make does not work when compiling libmpfr.so.4

zlin-ir, 2019-11-09:
Hi, first of all, thank you for the great work.
I am installing the tool on a new Raspberry Pi 4B following the instruction for bcm43430a1. 
At the step of compiling libmpfr.so.4, I met a problem when I execute ‘make’ with following information:

CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/pi/nexmon/buildtools/mpfr-3.1.4/missing aclocal-1.15 -I m4
/home/pi/nexmon/buildtools/mpfr-3.1.4/missing: line 81: aclocal-1.15: command not found
WARNING: 'aclocal-1.15' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make: *** [Makefile:425: aclocal.m4] Error 127
root@raspberrypi:/home/pi/nexmon/buildtools/mpfr-3.1.4# 
root@raspberrypi:/home/pi/nexmon/buildtools/mpfr-3.1.4#

maskof, 2019-11-10:
Hey ! you can install Autoconf 

`sudo apt-get install autoconf`
and then `sudo autoconf -f -i`

zlin-ir, 2019-11-10:
Thanks. Could you be more specific please?

maskof, 2019-11-10:
You need to install aclocal-1.15 using autoconf since you must be having aclocal-1.16 on your Raspberry PI 4B. you just need to install autoconf and then execute` autoconf -f -i` , that's what i did to solve the problem.

zlin-ir, 2019-11-10:
Ok, I tried autoreconf-f -i. It works now. thanks you very much.