seemoo-lab/nexmon_csi #13: I have some issues with makecsiparams.

U201213632, 2019-11-10:
After how to use the command makecsiparams. I already ran 'make', however it shows 'bash: makecsiparams: command not found'

maskof, 2019-11-10:
you can execute it by tapping `./makecsiparams [option]` in the same directory

kaustubhKVS, 2020-02-13:
Hey Buddy !! I faced the same problem too.
Mostly it will be a problem with the path variables of the system. i.e. BASH is not able to identify makecsiparams as a command and it is not able to find it. 

All you have to do is update your path variables for the makecsiparams command.
Easiest way to do this will be to run the following commands:
`sudo su`
`export PATH=$PATH:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/makecsiparams`
for RASPI4 and 3B+

Basically all you have to do is add the path where makecsiparams executable file is present, in my case the path is at /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/makecsiparams . You can change the path according to the location of makecsiparams executable file in your system.
To find where is the makecsiparams directory in your system run the following command in a fresh terminal i.e. when you are in home directory on terminal:
'find . -type d | grep makecsiparams`

For more information on updating path vairables visit: [https://opensource.com/article/17/6/set-path-linux](url)
For more information on finding directories visit: [https://superuser.com/questions/159290/finding-a-directory-in-a-linux-terminal/159292](url)

mestrelan, 2023-12-12:
> Hey Buddy !! I faced the same problem too. You'll face the same issue with nexutil command. Mostly it will be a problem with the path variables of the system. i.e. BASH is not able to identify makecsiparams as a command and it is not able to find it.
> 
> All you have to do is update your path variables for the makecsiparams command. Easiest way to do this will be to run the following commands: `sudo su` `export PATH=$PATH:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/makecsiparams` for RASPI4 and 3B+
> 
> Basically all you have to do is add the path where makecsiparams executable file is present, in my case the path is at /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/makecsiparams . You can change the path according to the location of makecsiparams executable file in your system. To find where is the makecsiparams directory in your system run the following command in a fresh terminal i.e. when you are in home directory on terminal: 'find . -type d | grep makecsiparams`
> 
> For more information on updating path vairables visit: https://opensource.com/article/17/6/set-path-linux For more information on finding directories visit: https://superuser.com/questions/159290/finding-a-directory-in-a-linux-terminal/159292

Hi! I have the same problem with nexutil command not found. Somebody could help me? I even try to make a path:
export PATH=$PATH:/home/allan/nexmon/utilities/nexutil/
I also try to use make in the path /nexmon/utilities/nexutil/ but not work. Any ideas?

jlinktu, 2023-12-13:
What is your target platform (the wifi chip you are targeting)?
What is your build platform (the system you are trying to build `nexutil` on)?

mestrelan, 2023-12-18:
> What is your target platform (the wifi chip you are targeting)? What is your build platform (the system you are trying to build `nexutil` on)?

Our build platform is bcm4366c0 (Asus RT-AC86U). I don't understand the first question about the target platform. I think that I could get CSI data with the router Asus RT-AC86U. Sorry, could you explain me please?

jlinktu, 2023-12-19:
_I don't think that your build platform is the `bcm4366c0`, that would be unnecessarily complicated. But I can derive from your answer that your target platforms are the `bcm4366c0` and Asus RT-AC86U._

- "target" typically refers to the specific system or platform for which a software application, program, or development project is intended. Or in other words, it is the environment where the compiled software is intended to run. In our case this can be one one of `bcm4339`, `bcm4358`, `bcm43455c0`, and `bcm4366c0` and their respective host devices, such as the Asus RT-AC86U.
- "build platform" in the context of software development refers to the environment or configuration in which a software project is compiled, linked, and built. In our case we recommend an Ubuntu or Raspberry Pi OS, but could be any system when correctly configured.

[Point **10** of the README section for the `bcm4366c0`](https://github.com/seemoo-lab/nexmon_csi?tab=readme-ov-file#bcm4366c0) gives instructions on how to build `nexutil` for the Asus RT-AC86U.

mestrelan, 2023-12-19:
> [Point **10** of the README section for the `bcm4366c0`](https://github.com/seemoo-lab/nexmon_csi?tab=readme-ov-file#bcm4366c0) gives instructions on how to build `nexutil` for the Asus RT-AC86U.

Thank you! Another doubt: I did point 10 of the README section for the bcm4366c0 and I am follow the Usage section and I am trying to do point 4 of the Usage section, but maybe when I turned off the computer and router I have to do point 10 of the README section for the bcm4366c0 again, isn't it?

Another doubt: When I use the nexutil command I have to be in the path /home/allan/nexmon/utilities/nexutil/ , isn't it?

jlinktu, 2023-12-20:
No, you don't have to repeat point 10. You install `nexutil` on the router at the following location `/jffs/nexutil`. The binary will stay there. Also after a reboot.

You are supposed to connect to the router via ssh and run `nexutil` on the router - NOT on your local machine!
```
# connect to your router via ssh
$ ssh admin@<your router's ip>
# run nexutil (interface is eth5 for 2.4GHz and eth6 for 5GHz on the Asus RT-AC86U)
$ /jffs/nexutil -Ieth6 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==
```

closed 2023-12-26 by jlinktu

mestrelan, 2024-01-10:
Thank you very much! College was closed for the end of the year break, but now I was able to continue.

Now I got the error:
__nex_driver_io: error ret=-1 errno=95

That is my terminal output:

admin@RT-AC86U-E208:/# ifconfig eth6 up
admin@RT-AC86U-E208:/# /jffs/nexutil -Ieth6 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWq
u6q7qrsAAAAAAAAAAAAAAAAAAA==
__nex_driver_io: error ret=-1 errno=95
admin@RT-AC86U-E208:/# /jffs/nexutil -Ieth6 -s500 -b -l34 -m+IBEQGIAgAAESIzRFWqu
6q7qrsAAAAAAAAAAAAAAAAAAA==
__nex_driver_io: error ret=-1 errno=95
admin@RT-AC86U-E208:/# ifconfig eth5 up
admin@RT-AC86U-E208:/# /jffs/nexutil -Ieth5 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWq
u6q7qrsAAAAAAAAAAAAAAAAAAA==
__nex_driver_io: error ret=-1 errno=95
admin@RT-AC86U-E208:/# /jffs/nexutil -Ieth5 -s500 -b -l34 m+IBEQGIAgAAESIzRFWqu6
q7qrsAAAAAAAAAAAAAAAAAAA==
nexutil: Too many arguments
Try `nexutil --help' or `nexutil --usage' for more information.
admin@RT-AC86U-E208:/# /jffs/nexutil -Ieth6 -s500 -b -134 -vm+IBEQGIAgAAESIzRFWq
u6q7qrsAAAAAAAAAAAAAAAAAAA==
/jffs/nexutil: invalid option -- '1'
Try `nexutil --help' or `nexutil --usage' for more information.
admin@RT-AC86U-E208:/#

jlinktu, 2024-01-10:
You have to load the modified `dhd.ko` first, also after a reboot. (point 7)

mestrelan, 2024-01-10:
> You have to load the modified `dhd.ko` first, also after a reboot. (point 7)

It might be a silly question, but how do I load dhd.ko? I found it in the /jffs folder, but I don't know how to load it.

Should I run 7. (Enter the created subdirectory nexmon_csi and run make install-firmware REMOTEADDR=<address of your rt-ac86u> to compile our firmware patch and install it on your RT-AC86U router.) ?

jlinktu, 2024-01-10:
This has been discussed in several issues already..

Yes, you could rebuild the patched firmware, copy it to the router, and load it, by running: `make install-firmware REMOTEADDR=<address of your rt-ac86u>` as of step 7.

But you can also just reload the already patched firmware once it resides on the router as `/jffs/dhd.ko` by running: `ssh admin@<address of your rt-ac86u> "/sbin/rmmod dhd; /sbin/insmod /jffs/dhd.ko"` (as can be seen from [`Makefile:367`](https://github.com/seemoo-lab/nexmon_csi/blob/fdb25ef0e4e1402e968bb644d4914ad1a3d0a84d/Makefile#L367)).