seemoo-lab/nexmon_csi #14: raspberry4 :issues at configure the extractor and enable monitor

Zhang-xie, 2019-11-11:
when I start to use this tool on a raspberry Pi4 , I  ran into the following issues:
1.
root@raspberrypi:/home/pi# nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQAAAQCcPc8fW4kAAAAAAAAAAAAAAAAAAAAAAAAAAA==
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received
( is there anything I need to do on the off-the-shelf AP?)

2.
root@raspberrypi:/home/pi# ifconfig wlan0 up
root@raspberrypi:/home/pi# iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
command failed: Operation not supported (-95)
( I can't enable the monitor on wlan0, even if I had pkill the wpa_applicant. Moreover, when I running: pkill wpa_applicant , raspbeery can't connect to the AP that I want to catch the csi from it)

salmafg, 2019-11-12:
I'm also having this problem with raspberry pi 3B+

salmafg, 2019-11-15:
I found a solution to the problem in a different repository. The configuration steps in the README file are very similar but some of them were missing. After following all the step from here: https://github.com/seemoo-lab/nexmon/blame/master/README.md#L73 I was able to solve the problem. Hope this helps!

princeulrich2, 2019-11-27:
I'm also having this problem with raspberry pi 4 B. I have checked the steps in the following repository as well but I still get the issue: https://github.com/seemoo-lab/nexmon/blame/master/README.md#L73

Junye97, 2019-12-09:
Hi there, 

I am running a Raspberry Pi 3B+ but i believe this might be helpful. I was also experiencing the same issue and it turned out that every time you restart the Pi you need to reinstall nexmon_csi and the firmware. 

Cheers

princeulrich2, 2019-12-12:
yes indeed, that worked for me

princeulrich2, 2019-12-15:
It is not necessary to reinstall the firmware after a reboot. The nexmon repository provides some guidelines on how to load the modified driver after a reboot, on the following page: https://github.com/seemoo-lab/nexmon.
Basically you can run the following commands once and you won't have to reinstall the firmware after a reboot:

`driver=$(modinfo brcmfmac | head -n 1)
driver=$(echo $driver | sed 's/^.\{'10'\}//')
mv $driver $driver.orig
driver_path=${driver%???????????}
cp /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_4.19.y-nexmon/brcmfmac.ko $driver_path
depmod -a`