seemoo-lab/nexmon_csi #15: CSI for management frames

codeceptsDE, 2019-11-12:
Hi,
thanks for the great work!
I have a few questions regarding CSI extraction.
I'm running a Raspberry Pi 4 and got both monitor mode as well as CSI extraction to work. My goal is to get CSI specifically for deauthentication frames.

1. From reading the Schulz dissertation I understand that in the process of capturing CSI, the content / payload of the frames is corrupted and no longer available, i.e. you can't capture a frame like in monitor mode and extract its CSI on the same device. Is that correct?
2. The byte `makecsiparams -b` is comparing against is the first byte I see in Wireshark after the RadioTap header in frames captured using monitor mode? So because for "QoS Data" frames this byte is 0x88, these are the frames being processed when i use `-b 0x88` (like in the example in your README)?
3. I'm having trouble capturing CSI of management frames, specifically deauthentication frames. Is CSI extraction triggered on _every_ frame, including management frames? If so, how would I filter for these frames? Neither 0xc0 for deauthentication, nor 0xb0 for authentication, 0xa0 for disassociation or 0xd0 for action frames seemed to work for me using the byte filter. I got these bytes as I just described using Wireshark and monitor mode.

Only not using any filter or using 0x88 as a filter got me any resulting UDP frames so far. As i can't see the frames corresponding to the CSI, I'm not able to tell whether anything other than "QoS Data" frames is triggering CSI extraction.
I'm using `/makecsiparams -e 1 -c 0x1002 -C 1 -N 1 -m MAC_OF_CLIENT,MAC_OF_ROUTER`.
Interestingly, I can capture deauthentication frames in monitor mode, but if I deauthenticate my device while using csi extraction mode, I capture no UDP packet whatsoever, even using no byte filter.

Can you please confirm weather my understanding is correct so far and point me in the right direction? Thanks!

codeceptsDE, 2019-12-17:
If anyone is having the same questions:
Regarding 3: I noticed many 802.11n access points tend to fall back to 802.11b for management frames, which is not using OFDM but CCK, and Nexmon can't get CSI for these. I got around this by using a 802.11ac access point, the fallback of which is 802.11a, which still uses OFDM.
1 & 2 seem to be correct.

Another interesting thing I observed:
The MAC filter `-m` only worked for the first MAC address, `MAC_OF_CLIENT` in my example, and `MAC_OF_ROUTER` has no effect as far as I could tell. In other words, capturing CSI only worked for one MAC address at a time. I am using a Raspberry Pi 4 running Raspbian with kernel 4.19.

closed 2019-12-17 by codeceptsDE

stepanmazokha, 2021-02-02:
@codeceptsDE thank you for discussing these issues! This question is a little late, but could you suggest any specific devices which are capable of emitting frames (in my case, probe requests) on 2.4 GHz using OFDM modulation?

codeceptsDE, 2021-02-03:
> @codeceptsDE thank you for discussing these issues! This question is a little late, but could you suggest any specific devices which are capable of emitting frames (in my case, probe requests) on 2.4 GHz using OFDM modulation?

@stepanmazokha As I said, I fixed my particular problem by switching to a 802.11ac access point, which is to say using 5GHz. But that is just because the AP I was using would downgrade 802.11n (2.4GHz OFDM frames) to 802.11b (2.4GHz CCK frames) for management frames, which the 802.11n standard allows (and 802.11ac does not, only down to 802.11a).
On the other hand, I don't see a reason why any 802.11n device would not be **able** to send any type of packet using any one of the modulations or any one of the modulations of its fallback standards, if you have control of the firmware. Afaik it's not like devices _have_ to downgrade, just that they _may_, probably to ensure more robust transmission at lower throughput. That being said, I myself have not done much frame injection, but I reckon even your Pi might be able to do that.

stepanmazokha, 2021-02-03:
@codeceptsDE thanks for the insight! I've had success injecting probe request frames on the 36th channel (5 GHz), and was able to capture them quite well using Nexmon CSI on RPi 4B+. And indeed, as you said, I've seen constant downgrading on 2.4 GHz on all of my attempts (both using Alfa AWUS036ACH which supports a/b/g/n and Archer C20 router).

For my research, I need to capture frames on both 2.4 and 5 GHz, and hence my hopes are that with Nexmon I can achieve OFDM modulation. That discussion continues [in this issue...](https://github.com/seemoo-lab/nexmon_csi/issues/178)

Thank you for your response!