seemoo-lab/nexmon_csi #16: CSI collection between 2 Asus rt-ac86u

mlo77, 2019-11-14:
Hello,

Thank you for this nice project.
My intention is to use 2 asus ac86u routers, one that will perform packet injection and the other CSI extraction. Your instructions for CSI extractions seem quite clear although I have not reach that stage yet since I am still to figure out how to perform packet injection.

1/ Could you confirm if the rt-ac86u is capable of injection ? At this time of writing it is not listed as one of the supported devices in your nexmon repository Readme.
2/ If yes, what method would do recommend ? Should I implement a user space app that makes ioctl commands as I read from issues posted here 2 years ago, or would there be now another tool that can do the job.

Thanks for reading and for any reply

matthiasseemoo, 2019-11-14:
If you are fine generating your frames in the firmware of the transmitters
Wi-Fi chip. Then, you can directly send them from there. We currently did
not publish the sendframe.c file for the bcm4366, but that would be the
easiest way to go and it would allow you to inject any frame you want and
you are very flexible to modify the modulation and bandwidth of the
transmitted frames.

Here, are some helper functions for the bcm4339:
https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4339/6_37_34_43/nexmon/src/sendframe.c

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

mlo77, 2019-11-18:
Thank you for the hint.

I've added some helpers functions into the bcm4366 folder, however I am having errors at linking stage :
`no memory region specified for loadable section ' .text.memset'`

If I restrain from calling memset in the code, then it's another function link error like text.hndrte_add_timer.
My guess is that I have to specify the function addresses in one of the linker scripts, like in gen/nexmon.ld. But how to find the function addresses ?

matthiasseemoo, 2019-11-18:
You need to specify the addressen kn the wrapper.c file and you need IDA or
another tool to reverse engineer the firmware and find the addresses.