seemoo-lab/nexmon_csi #17: some issues about nexmon_csi on Asus rt-ac86u

Zhang-xie, 2019-11-15:
hello,
Thank you for your amazing work!!!
When I set nexmon_csi on Asus rt-ac86u which is the recent version with BCM4366E, I run into the following issues. 
1) In the getting started guide, when I finished : make install-firmware REMOTEADDR=<address of your rt-ac86u>, I lost the connection with the router. I reboot the router and then connect to it gain.(I do not know if this is norm)
2) after following the getting started guide, I get some troubles in running : nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA== 
the running error:
admin@RT-AC86U:/jffs# ./nexutil -Ieth6 -s500 -b -l34 vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==
__nex_driver_io: error
( I don't know the meaning of ‘-s500’ )

Thanks for reading and reply!

jlinktu, 2019-11-15:
Hi, i guess you are connected to the router via wifi?
1. installing the patched firmware includes resetting the wifi chip, thus also resetting your current connection if you are connected to the router via wifi. you are supposed to be connected to it using a cable.
2. after rebooting the router the default firmware is loaded instead of the patched one, hence you won't be able to extract csi. (500 is the ioctl id but that has nothing to do with your problem)

so both problems should be solved by connecting to the router using a cable during the whole process

Zhang-xie, 2019-11-18:
Thank you very much @jlinktu !!
I tried it like what you say (using a cable to connect; do not reboot), while it still can't work. And there are more abnormal things happened as following:
1. after running :make install-firmware, the eth5 and eth6 are disappeared.
2. And then, I using : ifconfig eht6 up ;
3. finally, I run :./nexutil -Ieth6 -s500 -b -l34 vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==
then error like this:
__nex_driver_io: error

jlinktu, 2019-11-18:
After dhd.ko is reloaded (the firmware is installed), you should be able to bring up the interfaces by running for each of them:

```
wl -i <interface> up
wl -i <interface> radio on
ifconfig <interface> up
```

Zhang-xie, 2019-11-23:
Thank you for your kind reply @jlinktu  !!!!!
I have tried what you said, and it works !!! 
But I still want to ask you about some details on the install of this device, I hope you don't mind!

1. Based on the issues #8, I just need to install it in one dvices that I want to get csi from it. Dose this means this device(router) is the receiver? And I can get the csi which is send from another device such as the connected cellphone .( Am I right?)

2. After finish the Usage, it says csi data will send from port 5500(UDP)  on ip address 255.255.255.255. And then, dose it right to get a cable from my computer to connect the lan interface of this device to get the csi data?

3. Do I need to program with C on this device? If so, I want to kwon about the basic steps,  because I'm not good at this kind of programming.
Thanks again for your advice!!!!!!

ykoneee, 2019-11-23:
@jlinktu I'm @Zhang-xie 's colleague, sorry for so many questions.
After several experiments, our team has understood how to collect CSI data, the problem from @Zhang-xie  has been solved.
But we now have a new question that we hope will be answered. 
After running
```
wl -i <interface> up
wl -i <interface> radio on
ifconfig <interface> up
```
We can finally execute all the required commands in Usage without error. But we can't use our mobile phone to connect the WiFi because we did not find the target wifi in the mobile phone wifi list.
Then we chose to change the wifi to no password mode on the router control interface on the computer side via cable. After that we could find the target wifi in the phone wifi list, but still can't connect to it.The phone stuck at "Obtaining IP address".
how to solve this problem? thank you for your help!

BonnieLi1, 2020-01-15:
@jlinktu     @Zhang-xie  @ykoneee 
Hi, I met same problems here. I have run  [  ifconfig <interface> up ]. 
However, it still showed __nex_driver_io: error   when I run [  nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==  ].  So may I know how you make it work later?
Thanks for your help!!

BonnieLi1, 2020-01-15:
> Thank you very much @jlinktu !!
> I tried it like what you say (using a cable to connect; do not reboot), while it still can't work. And there are more abnormal things happened as following:
>
>     1. [...]

@jlinktu @Zhang-xie @ykoneee
Hi, I met same problems here. I have run [ ifconfig up ].
However, it still showed __nex_driver_io: error when I run [ nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA== ]. So may I know how you make it work later?
Thanks for your help!!

bachejames, 2020-03-23:
> After dhd.ko is reloaded (the firmware is installed), you should be able to bring up the interfaces by running for each of them:
> 
> ```
> wl -i <interface> up
> wl -i <interface> radio on
> ifconfig <interface> up
> ```

@jlinktu does this mean that we have to install Openwrt on the Asus router before all of this? As isn't `wl` an Openwrt command?

(Context: I am in the same positon of following the standard instructions and getting nexutil: command not found)

jlinktu, 2020-03-23:
`wl` is a command-line tool by broadcom and comes with the default firmware for the asus rt-ac86u.
Try to execute `nexutil` with the path to the binary, e.g.: `/jffs/nexutil`.

bachejames, 2020-03-28:
Many thanks for the reply @jlinktu, I'm afraid I am still struggling on this one.

My best guess is that I should be able to run the `wl` commands from any directory in Xubuntu, and that I can run them just as you stated, without any command on the front, however I am struggling to get it working. Below I list all the different things I have tried and the terminal response. Would you please be able to highlight which command is correct and why it is not working? Very many thanks!

- `wl -i enp0s3 up` command returns -> `wl: command not found`
- `/usr/sbin/wl -i enp0s3 up` (as you do a similar command in the readme) command returns -> `bash: /usr/sbin/wl: No such file or directory`
- (within utilities/nexutil dir) `./nexutil wl -i enp0s3 up` command returns -> `nexutil: Too many arguments`
- `/jffs/nexutil wl -i enp0s3 up` (as I think you were suggesting above) command returns -> `bash: /jffs/nexutil: No such file or directory`
- `ssh admin@192.168.1.1 wl -i enp0s3 up` (i.e. trying to run the command on the router via SSHing into it as [this](https://wiki.dd-wrt.com/wiki/index.php/Wl_command) suggests) command returns -> `sh: wl: not found

jlinktu, 2020-03-28:
As stated in the readme, all commands but the `makecsiparams` call shall be executed on the extraction device (the rt-ac86u in your case). Do not mix up `nexutil` and `wl` in one command, they are completely independent of each other.
I recommend you to first open a ssh connection:
`ssh admin@192.168.1.1`
After login, you should be able to run commands the way you are used to but on the router, e.g.:
`wl -i eth6 up`, `ifconfig eth6 up`, `wl -i eth6 monitor 1`
For `nexutil` you still need to include the full path: `/jffs/nexutil ...`, or alternatively add `/jffs` to the environment variable `PATH`.
Also make sure to use the correct interface. On the ac86u there are two WiFi interfaces, `eth5` for 2g band and `eth6` for 5g band.

bachejames, 2020-04-02:
> As stated in the readme, all commands but the `makecsiparams` call shall be executed on the extraction device (the rt-ac86u in your case). Do not mix up `nexutil` and `wl` in one command, they are [...]

Awesome! That's super useful, thanks a ton Jacob!

yujianyuanhaha, 2020-05-21:
> Hi, i guess you are connected to the router via wifi?
>
>     1. installing the patched firmware includes resetting the wifi chip, thus also resetting your current connection if you are connected to [...]

may i know by `using a cable`, we refer to a ethernet cable, or a USB cable.

Best.

hanabhp, 2020-08-11:
Hi @jlinktu

I really like this tool, great work!

I'm trying to install nexman_csi on the router. So far, I have finished the 'getting started' with no error.
Moving on, I have the following questions/issues:

- After installing nexmon_csi (using cable connection), the router SSIDs are disappeared (can't connect to the router through Wi-Fi), is this normal?!  

- When running `/jffs/nexutil eth5 -s500 -b -l34 -vm+IBEQGIAQAUGHebyhgAAAAAAAAAAAAAAAAAAAAAAAAAAA==` on the router, I get `too many arguments error`.
 any pointer is appreciated!

Mweme, 2021-08-18:
Hello Team, I have installed well nexmon in Rpi3 but I have failed to know how do I extract the CSI.
When I try to run this **root@raspberrypi:/home/pi/nexmon_csi/utils/makecsiparams# makecsiparams -c 157/80 -C 1 -N 1 -m b8:27:eb:74:37:d5 -b 0x88** I ended up with this error **bash: makecsiparams: command not found**

My file in nexmon_csi are
pi@raspberrypi:~/nexmon_csi $ ls
brcmfmac_4.19.y-nexmon  gfx       patch.ld   src
brcmfmac_5.4.y-nexmon   Makefile  README.md  utils
pi@raspberrypi:~/nexmon_csi/utils $ ls
makecsiparams  matlab

Any help will be appreciated

zeroby0, 2021-08-19:
You need to compile makecsiparams, it's in the utils folder, run `make` in the folder. And then copy the binary generated to be somewhere in your path.

nabeelni, 2022-09-13:
@Zhang-xie hi i am facing a similar problem to u installing the nexmon on asus,

I get the following error:
COPYING TO ROUTER dhd.ko => /jffs/dhd.ko
ssh: connect to host 192.168.1.1 port 22: Connection refused
lost connection
make: *** [Makefile:365: install-firmware] Error 1

Dont know if  i am missing something i followed the start guide!

closed 2026-01-29 by jlinktu