seemoo-lab/nexmon_csi #18: Tx side code for CSI collection

quantumhub, 2019-11-25:
Dear all, thank you for your amazing work!
It will be very helpful to have Tx side code for CSI collection. 

I followed the instruction on first page but then noticed that it is for receiver side only. I have not yet get any CSI.
It seems that it will only extract CSI from some very specific packet based on the instruction description (e.g. makecsiparams -c 157/80 -C 1 -N 1 -m 00:11:22:33:44:55,aa:bb:aa:bb:aa:bb -b 0x88), but not general any packet. (I am not sure if this understanding is correct?).

It is very attractive to get the 4x4 80MHz CSI mentioned in the paper, Free Your CSI: A Channel State Information Extraction Platform For Modern Wi-Fi Chipsets. Could you provide a full demo code (including Tx side code) for Ac86u?

Junye97, 2019-12-09:
Hi there, 

I am not sure what you meant by the Tx side and Rx side of things as I am also new to this, but I managed to collect some CSI data using a Raspberry Pi 3B+. 

To my understanding, you need to remove the -b parameter in the _makecsiparams_ command as it limits CSI to be collected with packets starting with 0x88. In addition, it seems that nexmon_csi has the wifi card running on monitor mode, thus the device might not be able to connect to any AP. I am not sure how to get CSI with MIMO as raspberry pi only has 1 antenna, but I suppose if your device supports MIMO you should be able to have 4x4 CSI. 

Cheers

SniperssSs, 2019-12-14:
> Hi there,
>
> I am not sure what you meant by the Tx side and Rx side of things as I am also new to this, but I managed to collect some CSI data using a Raspberry Pi 3B+.
>
> To my understanding, you need [...]

Hi mate, 

I guess the Tx and Rx he has mentioned means the router side and extraction device side. May I ask which kind of router are you using? And would it be necessary to set the wifi in a password-less mode?
Thanks. 

Best,

SniperssSs, 2020-02-07:
> Dear all, thank you for your amazing work!
> It will be very helpful to have Tx side code for CSI collection.
>
> I followed the instruction on first page but then noticed that it is for receiver side [...]

Do you mean the software-defined radios maybe?
Is this? https://github.com/seemoo-lab/mobisys2018_nexmon_software_defined_radio

wyhpolyu2020, 2020-06-18:
> Hi there,
>
> I am not sure what you meant by the Tx side and Rx side of things as I am also new to this, but I managed to collect some CSI data using a Raspberry Pi 3B+.
>
> To my understanding, you need [...]

dear, yes you are right. When I turn on the monitor mode, no AP could be accessed. So, how can I collect the csi data? How can I ping the extractor device using another laptop?

jlinktu, 2026-03-16:
https://github.com/seemoo-lab/nexmon_tx_task

closed 2026-03-16 by jlinktu