seemoo-lab/nexmon_csi #19: Protocol not supported error with Raspberry pi 4B

princeulrich2, 2019-11-28:
I have followed the steps to install the patch on a new raspberry pi 4 B (bcm43455c0). However when I try to configure the extractor with nexutil I get the following errors:

```
root@raspberrypi4B:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/makecsiparams# ./makecsiparams -c 36/80 -C 1 -N 1 -m 58:ef:68:48:5f:c3 -b 0x88
KuABEQGIAQBY72hIX8MAAAAAAAAAAAAAAAAAAAAAAAAAAA==
```

```
root@raspberrypi4B:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/makecsiparams# nexutil -Iwlan0 -s500 -b -l34 -vKuABEQGIAQBY72hIX8MAAAAAAAAAAAAAAAAAAAAAAAAAAA==
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received
```

```
root@raspberrypi4B:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/makecsiparams# sudo nexutil -k36/80 && nexutil -k
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: socket error (93: Protocol not supported)
nex_init_netlink: bind error (9: Bad file descriptor)
nex_init_netlink: connect error (9: Bad file descriptor)
ERR (__nex_driver_netlink): no valid answer received
chanspec: 0x6863, 85/160
```

I also get the errors if I simply run nexutil -k as shown above. I am not sure what is going wrong, your help will be very much appreciated

zlin-ir, 2019-12-03:
Hi @princeulrich2 
Have you solved your problem?
It happened to me as well. I am using RPI 4B.

Junye97, 2019-12-06:
Hi there, 

Just try to remake and reinstall the firmware and nexutils following the Get Started section. It worked for me. You can refer to [this issue](https://github.com/seemoo-lab/nexmon/issues/67).

Cheers

princeulrich2, 2019-12-12:
yes indeed, I solved the problem by reinstalling the firmware after every reboot. It turns out that the default firmware is loaded after every reboot, instead of the patch

princeulrich2, 2019-12-15:
Actually we don't have to reinstall the firmware after a reboot. I just found that the nexmon repository provides some guidelines on how to load the modified driver after a reboot, on the following page: https://github.com/seemoo-lab/nexmon.
Basically I created the following script with the commands which were provided on the nexmon page and ran it once. Now I don't have to reinstall the firmware after a reboot

`driver=$(modinfo brcmfmac | head -n 1)
driver=$(echo $driver | sed 's/^.\{'10'\}//')
mv $driver $driver.orig
driver_path=${driver%???????????}
cp /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/brcmfmac_4.19.y-nexmon/brcmfmac.ko $driver_path
depmod -a`

closed 2020-01-24 by princeulrich2