seemoo-lab/nexmon_csi #20: parameters setting in makecsiparams

SniperssSs, 2019-12-08:
Hi there, 

A beginner's question, the first step after 'get started', in the example
`makecsiparams -c 157/80 -C 1 -N 1 -m 00:11:22:33:44:55,aa:bb:aa:bb:aa:bb -b 0x88
m+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==`
What is the string `m+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==`?

Thanks in advance.

matthiasseemoo, 2019-12-08:
Did you read the readme?

quantumhub, 2019-12-08:
@SniperssSs It is used in step 4. nexutil will use this enocded string as command.

> Hi there,
>
> A beginner's question, the first step after 'get started', in the example
> `makecsiparams -c 157/80 -C 1 -N 1 -m 00:11:22:33:44:55,aa:bb:aa:bb:aa:bb -b 0x88 [...]

SniperssSs, 2019-12-09:
> […](#)

Thank you for your reply. That is the example in the ReadME file.

SniperssSs, 2019-12-09:
> @SniperssSs It is used in step 4. nexutil will use this enocded string as command.

Thank you for your reply. But how can I determine this in step 1?

jlinktu, 2019-12-09:
It's the output generated by makecsiparams.

SniperssSs, 2019-12-09:
> It's the output generated by makecsiparams.

LoL) Thanks

closed 2019-12-09 by SniperssSs