seemoo-lab/nexmon_csi #21: Extracting CSI without MAC Address

Junye97, 2019-12-10:
Hi there, 

It seems that nexmon_csi requires the MAC address of the devices in order to get the CSI. Is there a way to get around that and extract CSI without any MAC address required, i.e. extracting all the CSI packets from a specific channel? I am digging into the source code but haven't made much progress. 

Cheers

jlinktu, 2019-12-10:
Hi,
https://github.com/seemoo-lab/nexmon_csi/blob/8260f5b7a8617ed064f561b972b66c535ed49329/src/ioctl.c#L133
just set the number of mac addresses to filter for to 0.

Junye97, 2020-01-28:
Hi jlinktu, 

Thank you for your reply. However, if I set the `params->n_mac_addr = 0;` as follows:

[image]

there will be no CSI packet captured. Please advise. 

Cheers

zeroby0, 2020-01-28:
There is another attribute above "use_pkt_filter", try setting that to 0. I'm about to test in 1 hr, will let you know if that works.

jlinktu, 2020-01-28:
There is an error in the ucode code preventing csi extraction when no mac address is given. I try to push a fix for that soon.

Junye97, 2020-01-29:
Thank you very much @jlinktu.

zeroby0, 2020-01-29:
@jlinktu  Please do, I need it for my thesis. Thanks :D

jlinktu, 2020-01-29:
Done.
If _n_mac_addr_ is set to 0, address filtering is disabled.
If you use `makecsiparams` to configure the extractor you can set this by simply leaving out the _-m_ parameter.

Using this for the bcm43455c0 and bcm4358 also requires updating your main nexmon repository!

Junye97, 2020-01-30:
Hi @jlinktu, thank you so much for your effort. It works for me now and I'm closing this issue. 

Cheers

closed 2020-01-30 by Junye97