seemoo-lab/nexmon_csi #22: Need help starting the usage

zlin-ir, 2019-12-11:
Hi guys,
I am trying to extract CSI with a Raspberry Pi 4B (4.19.75-v7l+).

By using nexutil -k157/80, I set the channel spec to 157/80.
By using iw wlan0 scan, I learn that the mac address to my router is 04:d9:f5:d9:06:38.
So, I get CSI parameter by using: makecsiparams -c 157/80 -C 1 -N 1 -m 04:d9:f5:d9:06:38. (I did not use the filter here).
Then, pkill wpa_supplicant
Then, ifconfig wlan0 up
Then, nexutil -Iwlan0 -s500 -b -l34 -vm+(generated CSI parameters).
Then, iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
ifconfig mon0 up
Finally, tcpdump -i wlan0 dst port 5500.

The result is everything looks fine during the above steps but I cannot receive/capture any packets.
Can anyone help? Thanks

SniperssSs, 2019-12-12:
I have the same problem. It tells:
`tcpdump: verbose output suppressed, use -v or -vv for full protocol decode listening on wlan0, link-type EN10MB (Ethernet), capture size ... bytes
[blank]`

Any idea? Thanks.

zlin-ir, 2019-12-13:
@SniperssSs same here.

maskof, 2019-12-13:
Hey, have you tried looking for other channel such as channel 100 or set multiple router mac addresses ?

princeulrich2, 2019-12-14:
Hey, I had a similar issue but I solved it by setting my router to the same channel from which I was trying to collect csi. In the wireless section of your router's settings you can set a specific channel. maybe it helps?

baozhongxu, 2019-12-23:
How can I change the sampling frequency to 100