seemoo-lab/nexmon_csi #24: WIFI 5G SSID missing after install firmware

zlin-ir, 2019-12-13:
Hi guys,

I am using Raspberry PI 4 Model B.
Not sure if this issue has happened to you as well that, after running make install-firmware to compile the nexmon_csi firmware patch, my RPI can no longer find any WIFI 5G SSID, which was connected before running this step. I have re-installed the whole system but this problem happened again.

Anyone knows the reason why this happening?

Thanks

SniperssSs, 2019-12-15:
Same here. So I cannot set the wifi channel same to the csi collected one i.e. 157 [cry]

SniperssSs, 2019-12-15:
But actually, I have lost all the wireless connections after `pkill wpa_supplicant` in the usage

zlin-ir, 2019-12-16:
@SniperssSs You can find some WIFI SSID via iw wlan0 scan except for those 5Gs.

zeroby0, 2019-12-17:
If you ran `apt-get remove wpasupplicant`, that removes wpasupplicant and aps dont show up. Instead, use `iw wlan0 scan`.

zlin-ir, 2019-12-18:
@zeroby0 I have successfully managed to listen to the WIFI info via tcpdump. But it stopped working after a while (about 20 minutes). And I use `iw wlan0 scan` again, it outputs _connection timed out_. Did you have this issue?

SniperssSs, 2019-12-18:
> @zeroby0 I have successfully managed to listen to the WIFI info via tcpdump. But it stopped working after a while (about 20 minutes). And I use `iw wlan0 scan` again, it outputs _connection timed out_. Did you have this issue?

Could you please share how you solved the previous problems?

zlin-ir, 2019-12-18:
> > @zeroby0 I have successfully managed to listen to the WIFI info via tcpdump. But it stopped working after a while (about 20 minutes). And I use `iw wlan0 scan` again, it outputs _connection timed out_. Did you have this issue?
> 
> Could you please share how you solved the previous problems?

I used RPI 3B+ instead, problem solved😂

zlin-ir, 2019-12-18:
> > @zeroby0 I have successfully managed to listen to the WIFI info via tcpdump. But it stopped working after a while (about 20 minutes). And I use `iw wlan0 scan` again, it outputs _connection timed out_. Did you have this issue?
> 
> Could you please share how you solved the previous problems?

Also, you need to change the router's channel and bandwidth on the router's control panel, say channel 157 and bandwidth 80MHZ

SniperssSs, 2019-12-18:
Thanks a lot. ( Lol @zlin-ir

closed 2019-12-19 by zlin-ir