seemoo-lab/nexmon_csi #25: TCPDUMP stopped working after running two minutes.

zlin-ir, 2019-12-19:
Hi guys,

I believed that I have installed everything correctly and my device (RPI 3B+) started listening via tcpdump. I have also recorded the info to a .pcap file and read it with provided Matlab code.

However, I notice that tcpdump stopped working every time after running two minutes and thirty seconds. And when I run `iw wlan0 scan` again, it gives 

> command failed: Connection timed out (-110)

or when I run 'iwlist wlan0 scan', it gives

> Interface doesn't support scanning : connection timed out

Has anyone have the same issue?
Thanks

Junye97, 2020-01-07:
Hi there, 

Am not sure why your setup stops to work after exact 150 seconds. From what I have experienced, there seem to be some kind of buffering which will be filled up and the tcpdump will stop collecting CSI packets. My walk around is to restart the R Pi every time after each collection. 

Also, have a look at the original nexmon GitHub page as it mentions something about how to exit the monitor mode for the wlan0 interface. I am not sure if you would be able to run iw scan when the hardware is in monitor mode. 

Cheers

zlin-ir, 2020-02-08:
> Hi there,
>
> Am not sure why your setup stops to work after exact 150 seconds. From what I have experienced, there seem to be some kind of buffering which will be filled up and the tcpdump will stop [...]

Thanks for your reply @Junye97 .
It was not exactly 150 seconds but it was some number around this. 
Is it possible to clear up the buffer without restating the RPI as I am using it for a localization system and I need it to run continuously.

mzakharo, 2020-02-08:
I am having the same problem on rpi4. My workaround now is to unload and reload the  firmware. Does anyone know what part of the code is misbehaving?

mzakharocsc, 2020-02-09:
Ok, I uninstalled bluetooth from user/kernel space, not sure if this did anything, but it seems like it can go on for longer before breaking.

After running for hours, my RP4 stopped receiving anything  on UDP 5500 again. The only thing I see in the dmesg log is:

`brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted`

A workaround is to `modprobe -r brcmfmac`, then `modprobe brcmfmac` and then run `make install-firmware`

Based on dmesg log, the error seems to occur inside the firmware itself, so does anyone know whether there are debug tools to help resolve this?

mzakharo, 2020-02-11:
This may be related to #41 and #39