seemoo-lab/nexmon_csi #26: problem with command "make install-firmware"

WangPanHUST, 2019-12-29:
Thank you for your great work!
I'm doing the prepared work following the instruction for nexus6p(bcm4358).
When I  run make install-firmware in nexmon_csi,  there is a problem:

Assembler messages:
Fatal error: invalid -march= option: 'armv7-r'

I found the original definition ‘NEXMON_ARCH=armv7-r’ in the file definitions.mk.
The anexus6p is actually armv8l architecture. How to solve this problem?

Looking forward to your answer. Thank you.

jlinktu, 2019-12-31:
You might use a wrong compiler.
Have you sourced the setup_env.sh file in the nexmon root directory before building?

WangPanHUST, 2020-01-03:
Thank you for your reply! I have solved this problem.

There is another problem when i extracting CSI. After i run "nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==", it showes "_nex_driver_io:error".

The system of my nexus6p is Android 7.1.2. Can you give me some suggestions? 
Thank you.

WangPanHUST, 2020-01-06:
You must first run “ifconfig wlan0 up” to open the wlan0. 
The deault name of nexus 6p is wlan0!

closed 2020-01-06 by WangPanHUST