seemoo-lab/nexmon_csi #27: The length of UDP packets is wrong

xcl0416, 2019-12-31:
Sorry to bother you. I have followed the instruction and log the UDP packets with Nexus 5 like this.  
[image]
However, when I process the pcap file with MATLAB program, I cannot find CSI from it. I find that the length of UDP packet of my pcap file is different from yours, which is 1034. I have no idea about this. Could you please tell me why.

jlinktu, 2019-12-31:
Adapt https://github.com/seemoo-lab/nexmon_csi/blob/8260f5b7a8617ed064f561b972b66c535ed49329/utils/matlab/csireader.m#L13 to 20, the bandwidth in MHz you used during extraction. You also want to change https://github.com/seemoo-lab/nexmon_csi/blob/8260f5b7a8617ed064f561b972b66c535ed49329/utils/matlab/csireader.m#L12 to _bcm4339_ for processing captures made with a Nexus 5.

xcl0416, 2020-01-01:
It works, thank you!

closed 2020-01-01 by xcl0416