seemoo-lab/nexmon_csi #28: make install-firmware: Assembler ERROR:   No %arch defined

quantumhub, 2020-01-02:
I am using 
bcm4358 | 7_112_300_14_sta | Nexus 6P

When I  run make install-firmware, I get following error. Could anyone help? Thanks!

"  COLLECTING STATISTICS read /home/XXX/Documents/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  ASSEMBLING UCODE src/csi.ucode.bcm4358.7_112_300_14_sta.asm => gen/ucode.bin
gcc: error trying to exec 'cc1plus': execvp: No such file or directory
Assembler ERROR:
  No %arch defined
Makefile:211: recipe for target 'gen/ucode.bin' failed
make: *** [gen/ucode.bin] Error 1
"

WangPanHUST, 2020-01-02:
find the definitions.mk in line 3 of this makefile. 
check the parameter "arch".

quantumhub, 2020-01-02:
Thanks for @WangPanHUST 's reply. More detailed instruction could be very helpful!

From the line 3 of this makefile:
"include ../version.mk
include $(FW_PATH)/definitions.mk"

FW_PATH=$(NEXMON_ROOT)/firmwares/bcm4358/7_112_300_14_sta

I found the definitions.mk in the directory "nexmon/firmwares/bcm4358/7_112_300_14_sta":
NEXMON_ARCH=armv7-r

Please suggest how to fix the issue as next step? Thanks.

> find the definitions.mk in line 3 of this makefile.
> check the parameter "arch".

WangPanHUST, 2020-01-03:
Hi,
I think there are two possible solutions for you.

1. "No %arch defined", it seems like a file format error. Are you clone the 'nexmon' repository directly on your Linux computer? If you copied it from a windows computer, there will be a lot of probelms in the next steps. All steps are ok after I re-cloned this repository on my linux computer. You'd better to check the file format of the definitions.mk.

2.You can find gcc-arm-none-eabi-5_4-2016q2-linux-x86 in /nexmon/buildtools/.
If the file format is ok, you can download the "gcc-arm-none-eabi-5_4-2016q2" in the official website. 
Then, change the PATH of $CC to this official version by using command "export $CC=/**". 
The original definition of $CC exist in the setup_env.sh.

Hope this helps.(I think the high probability is cause 1)

quantumhub, 2020-01-05:
Thanks for the detailed instructions!
I am on Xubuntu. So, it is unlikely due to cause 1.
For cause 2, I google the error message and find that installation of g++ can solve the problem:
sudo apt-get install g++

I will further verify if it solves the issue, since it is expecting a cross-compiling.

jlinktu, 2020-01-05:
Have you sourced the setup_env.sh script before?

quantumhub, 2020-01-05:
> sourced the setup_env.sh

I have sourced setup_env.sh. The error prompt vanishes after "sudo apt-get install g++".