seemoo-lab/nexmon_csi #29: collecting CSI

U201213632, 2020-01-02:
Can I use commodity wireless router as the transmitter, when I collect CSI by using raspberypi. How to set the channel parameter, i.e., the channel index and the bandwidth as the router sets the channel and bandwidth automatically?

quantumhub, 2020-01-02:
> Can I use commodity wireless router as the transmitter, when I collect CSI by using raspberypi. How to set the channel parameter, i.e., the channel index and the bandwidth as the router sets the channel and bandwidth automatically?

You can use a regular wifi card to scan and find the channel and BW parameters of your target router.

U201213632, 2020-01-03:
> You can use a regular wifi card to scan and find the channel and BW parameters of your target router.

thanks