seemoo-lab/nexmon_csi #30: problem with Configure the extractor using nexutil:'_nex_driver_io:error'

WangPanHUST, 2020-01-03:
I'm using nexus6p(Android 8.0.0) to collecting csi. In the step 4 of Usage : Configure the extractor using nexutil and the generated parameters. After i run command:
nexutil -Iwlan0 -s500 -b -l34 -vxxxx...
it showes '_nex_driver_io:error'.

How to solve this problem? 

Looking forward to your reply.Thank you!

jlinktu, 2020-01-05:
Have you loaded the patched firmware before?

quantumhub, 2020-01-05:
> I'm using nexus6p(Android 8.0.0) to collecting csi. In the step 4 of Usage : Configure the extractor using nexutil and the generated parameters. After i run command:
> nexutil -Iwlan0 -s500 -b -l34 [...]

One possible cause is that you have not yet brought wlan0 up, you can try:
ifconfig wlan0 up

WangPanHUST, 2020-01-06:
Thanks for your instruction!
I used a wrong name of interface by "dummy0". Acctually, the default interface nam of nexus 6p is "wlan0".

closed 2020-01-06 by WangPanHUST