seemoo-lab/nexmon_csi #31: no CSI packet observed by tcpdump

quantumhub, 2020-01-05:
I tried nexutil and tcpdump on Nexus 6P. However, no CSI packet observed by tcpdump. Could anyone help?

My router is in 11a/n/ac mixed mode on channel 149/80. I have an another phone communicating with this router. So, CSI is supposed to captured on this channel:

Xubuntu:~$ adb shell
angler:/ $ su

makecsiparams$ ./makecsiparams -c 149/80 -C 1 -N 1
m+ABEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==

angler:/ # nexutil -Iwlan0 -s500 -b -l34 -vm+ABEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA== 
angler:/ # nexutil -Iwlan0 -m1
angler:/ # iw dev
phy#0
	Unnamed/non-netdev interface
		wdev 0x3
		addr 26:xx
		type P2P-device
	Interface wlan0
		ifindex 5
		wdev 0x1
		addr 24:xx
		type managed
angler:/ # tcpdump -i wlan0 dst port 5500
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel
angler:/ #

jlinktu, 2020-01-05:
Check by running ```$ nexutil -k``` if the chanspec get's set to 149/80.

quantumhub, 2020-01-05:
> Check by running `$ nexutil -k` if the chanspec get's set to 149/80.

Indeed that wlan0 received packets, but it is strange that no CSI has been obtained.
nexutil -k shows that Nexus 6P is watching on channel 149:

angler:/ # nexutil -k                                                          
chanspec: 0xd095, 149
angler:/ # tcpdump -i wlan0 dst port 5500                                      
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel
angler:/ # nexutil -Iwlan0 -m                                                  
monitor: 1

angler:/ # ifconfig
wlan0     Link encap:Ethernet  HWaddr 24:xx
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:38977 errors:0 dropped:25232 overruns:0 frame:0 
          TX packets:28 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:8148593 TX bytes:4892 

angler:/ # ifconfig           
wlan0     Link encap:Ethernet  HWaddr 24:xx
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:38980 errors:0 dropped:25235 overruns:0 frame:0 
          TX packets:28 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:8148857 TX bytes:4892

quantumhub, 2020-01-06:
> Check by running `$ nexutil -k` if the chanspec get's set to 149/80.

@jlinktu Thanks for the reply. The problem is not yet solved:

angler:/ # nexutil -Iwlan0 -k149/80                                            
angler:/ # nexutil -Iwlan0 -k                                                  
chanspec: 0xe09b, 149/80
angler:/ # tcpdump -i wlan0 dst port 5500                                      
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel
angler:/ # nexutil -Iwlan0 -w                                                  
__nex_driver_io: error
version: 0
length: 0
txframe: 0 (tx data frames)
txbyte: 0 (tx data bytes)
txretrans: 0 (tx mac retransmits)
txerror: 0 (tx data errors (derived: sum of others))
...

jlinktu, 2020-01-06:
Please try if it works when you use the address filter, there seems to be an error preventing extraction when 0 addresses are given. I will push a fix within the next days.

quantumhub, 2020-01-07:
> Please try if it works when you use the address filter, there seems to be an error preventing extraction when 0 addresses are given. I will push a fix within the next days.

Thanks for the hint!
It seems to be working after I add one MAC addr in the filter:)
./makecsiparams -c 149/80 -C 1 -N 1 -m 9x:XX

I will further check if content with length 1042 is CSI or not.

09:15:59.605172 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
09:15:59.609400 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
09:15:59.643422 IP 10.10.10.10.5500 > 255.255.255.255.5500: UDP, length 1042
^C
1046 packets captured
1046 packets received by filter
0 packets dropped by kernel

WangPanHUST, 2020-01-10:
There is another possible cause.
As the Nexus 6P doesn't connect to the router, the packets captured by the Nexus 6P are transmitted by the router to other devices which connect to the router.
If it showes no packets captured or the speed is slow, try to get the router to send data,like opening a web page.

yujianyuanhaha, 2020-05-26:
@quantumhub may i know the path of command `tcpdump`? i honest have not found it at RT-AC86 router. thanks