seemoo-lab/nexmon_csi #32: make[3]: /buildtools/ucode_extractor/ucodeext: Command not found

coldcoldmilk, 2020-01-10:
Hello, first of all, thank you for your work.
This is the error message I get:
```
  EXTRACTING UCODE
/buildtools/ucode_extractor/ucodeext -r bcmdhd_sta.bin -b 0x2D830 -l 0x8B82 -o u     code.bin
make[3]: /buildtools/ucode_extractor/ucodeext: Command not found
make[3]: *** [Makefile:7: ucode.bin] Error 127
make[3]: Leaving directory '/home/pi/Downloads/nexmon/firmwares/bcm4330/5_90_100     _41_sta'
make[2]: *** [Makefile:8: 5_90_100_41_sta/] Error 2
make[2]: Leaving directory '/home/pi/Downloads/nexmon/firmwares/bcm4330'
make[1]: *** [Makefile:7: bcm4330/] Error 2
make[1]: Leaving directory '/home/pi/Downloads/nexmon/firmwares'
make: *** [Makefile:5: firmwares] Error 2

```
What should I do？Thank you

And in your readme bcm43455c0,
`cd buildtools/isl-1.10`and `cd buildtools/mpfr-4.1.4`
I can't do this.because there are no isl-1.10 and mpfr-4.1.4, only isl-0.10 and mpfr-3.1.4.

jlinktu, 2020-01-10:
The wrong versions in the readme were just typos.
Follow the readme again and use isl-0.10 and mpfr-3.1.4.

coldcoldmilk, 2020-01-10:
Yes,i did.
According to your readme ,I use Raspberry Pi B4 and running Raspbian with kernel 4.19,and all is well.
Until in step 6  run `make` and then the mistake happened.

coldcoldmilk, 2020-01-10:
> The wrong versions in the readme were just typos.
> Follow the readme again and use isl-0.10 and mpfr-3.1.4.
Yes,i did.
According to your readme ,I use Raspberry Pi B4 and running Raspbian with kernel 4.19,and all is well.
Until in step 6 run make and then the mistake happened.

jlinktu, 2020-01-10:
Have you executed ```source setup_env.sh``` from the nexmon root directory before ```make```?
It sets some env variables required by some Makefiles in order to find i.a. ucodeext.

coldcoldmilk, 2020-01-11:
> Have you executed `source setup_env.sh` from the nexmon root directory before `make`?
> It sets some env variables required by some Makefiles in order to find i.a. ucodeext.

I made a stupid mistake. Thank you for your patience.I used this code `sh setup_env.sh` before.Thank you again.

closed 2020-01-19 by coldcoldmilk