seemoo-lab/nexmon_csi #33: Why use frames starting with 0x88?

zeroby0, 2020-01-12:
> From the readme:
``` makecsiparams -c 157/80 -C 1 -N 1 -m 00:11:22:33:44:55,aa:bb:aa:bb:aa:bb -b 0x88
m+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==
```

The `-b` flag filters out packets without `0x88`. So:
- why use frames with `0x88`? Are these beacon frames?
- I can capture a lot more packets without the filter. Is the data in those packets invalid? Because they're directed at a device in the network and not general beacon broadcasts.

Thanks in advance :)

jlinktu, 2020-01-12:
Hi,

as described in https://github.com/seemoo-lab/nexmon_csi/issues/3#issuecomment-541309501, the first byte of a frame contains version, type, and subtype fields of the frame control field. Using -b with 0x88, only csi of qos data frames would be extracted. You can use 0x80 for csi of beacon frames. If you want csi of any kind of frame you can also omit the filter. Sometimes it can be interesting to have csi only for specific frames.

The csi is not calculated by the mac layer content of the frame but a training sequence in the preamble. No specific frame type or additional measurement data required.

Reading csi does not corrupt the packet data. When using the filter, csi is only extracted for matching frames, that is when version, type, and subtype fields match with the filter byte.

closed 2020-01-19 by zeroby0

Mweme, 2021-08-18:
Hello team,
I'm new to the channel thing, but currently is the place I'm working my research.
I have installed nexmon in RPi and was successful and I have configured another RPi to work as AP.
But I would like to have guidance on the following:-

1) How do I capture CSI frames?
2) How do I visualize these frames?
3) Which file do these CSI frames are kept in Rx?
4) How analysis has been done?

Any assistance will be much appreciated.

zeroby0, 2021-08-18:
Why are you asking here though? Create a new issue, it will be helpful to others as well :D
@Mweme