seemoo-lab/nexmon_csi #34: Some problems in usage part on RT-AC86U

BonnieLi1, 2020-01-17:
Hi,
When I operated on Asus rt-ac86u as instruction suggests, I met the following problems.
1. After interface is up,  running
`admin@RT-AC86U:/jffs# ./nexutil -Ieth6 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7q
rsAAAAAAAAAAAAAAAAAAA==`
the error shows:
__nex_driver_io: error
However, when running the code without -s500, the error disappears. I have no idea...

2. Does anyone know if I need to download the tcpdump in RT-AC86U? Since it shows 
-sh: tcpdump: not found

Thanks for kindly reading and reply!!!!!

jlinktu, 2020-01-17:
Hi,

1. make sure the patched dhd.ko is loaded, `dmesg` should show a line containing `10.10.122.20 (nexmon.org/csi`
when you omit _-s500_, nexutil won't do anything -> no error

2. one option is to cross compile tcpdump for the rt-ac86u from the https://github.com/RMerl/asuswrt-merlin.ng repository using the https://github.com/RMerl/am-toolchains toolchain, tested on a x86_64 linux host:
  a) clone the am-toolchains repo and do exports and path settings according to the readme instructions for bcm-hnd
  b) clone the asuswrt-merlin.ng repo, change directory to `release/src/router/tcpdump-4.4.0`, then:
   ```
   CC=aarch64-linux-gcc AR=aarch64-linux-ar RANLIB=aarch64-linux-ranlib AS=aarch64-linux-as LD=aarch64-linux-ld NM=aarch64-linux-nm ac_cv_linux_vers=2 CFLAGS=-static CPPFLAGS=-static LDFLAGS=-static ./configure --host=x86_64-unknown-linux-gnu && make
   ```

BonnieLi1, 2020-01-21:
> Hi,
>
>     1. make sure the patched dhd.ko is loaded, `dmesg` should show a line containing `10.10.122.20 (nexmon.org/csi`
>        when you omit _-s500_, nexutil won't do anything -> no error
>
>     2. [...]
Thanks for your kindly  replying!!!!   @jlinktu 
According to your reply, I run `dmesg`. However, it showed the following message:
> dhd_prot_ioctl: status ret value is -4 
CONSOLE: 346239.493 wl0: wlc_scan_request_ex, can not scan while driver is down
CONSOLE: 346239.493 wl0: wlc_scan_request_ex, can not scan due to error -4
May I know how to fix it?
Thanks a lot!!!

jlinktu, 2020-01-21:
Also make sure the interface you want to use is up:
```
wl -i <interface> up
wl -i <interface> radio on
ifconfig <interface> up
```

BonnieLi1, 2020-01-22:
Thanks for your reply again!!! @jlinktu 
Now it did not display the driver error anymore!!
However, when I cross compile tcpdump and operate as you said, it shows the error:

> configure: error: C compiler cannot create executables

> See `config.log' for more details

And log displays:

> ./configure: line 2704: aarch64-linux-gcc: command not found

Since I haven't done this kind of compiling before, I have no idea about it.
Thanks a lot!!! Sorry for taking your time!!!

jlinktu, 2020-01-22:
This is basically a repetition of my first response but with all required commands easy to follow.
From a directory of your choice clone the toolchain:
```
git clone https://github.com/RMerl/am-toolchains.git
```
add the toolchain to your environment (this is only temporary):
```
export PATH=$PWD/am-toolchains/brcm-arm-hnd/crosstools-aarch64-gcc-5.3-linux-4.1-glibc-2.22-binutils-2.25/usr/bin:$PATH
export LD_LIBRARY_PATH=$PWD/am-toolchains/brcm-arm-hnd/crosstools-aarch64-gcc-5.3-linux-4.1-glibc-2.22-binutils-2.25/lib:$LD_LIBRARY_PATH
```
Proceed by cloning the source code and compiling libpcap (forgot about that earlier) and tcpdump (step 2b from my first response):
```
git clone https://github.com/RMerl/asuswrt-merlin.ng.git
cd asuswrt-merlin.ng/release/src/router/libpcap
CC=aarch64-linux-gcc AR=aarch64-linux-ar RANLIB=aarch64-linux-ranlib AS=aarch64-linux-as LD=aarch64-linux-ld NM=aarch64-linux-nm ./configure --host=x86_64-unknown-linux-gnu --with-pcap=linux && make
cd ../tcpdump-4.4.0
CC=aarch64-linux-gcc AR=aarch64-linux-ar RANLIB=aarch64-linux-ranlib AS=aarch64-linux-as LD=aarch64-linux-ld NM=aarch64-linux-nm ac_cv_linux_vers=2 CFLAGS=-static CPPFLAGS=-static LDFLAGS=-static ./configure --host=x86_64-unknown-linux-gnu && make
```

289536718, 2020-04-27:
@jlinktu 
Excuse me, sir. I also encountered the above problems when installing Nexmon_csi on Asus-Ac86u.

When I input: 

`CC = aarch64 Linux GCC ar = aarch64 Linux ar ranlib = aarch64 Linux as LD = aarch64 Linux LD nm = aarch64 Linux nm. / configure -- host = x86 ʄ unknown Linux GNU -- with pcap = Linux & & make`
The following prompt appears on the command line:
```
configure: WARNING: If you wanted to set the --build type, don't use --host.
If a cross compiler is detected then cross compile mode will be used.
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking target system type... x86_64-unknown-linux-gnu
checking for x86_64-unknown-linux-gnu-gcc... aarch64-linux-gcc
checking for C compiler default output file name...
configure: error: C compiler cannot create executables
See `config.log' for more details.

```
I guess I may be missing something, but there is no way to solve it. Now I upload the config file, please help me to have a look, thank you.

[config.log](https://github.com/seemoo-lab/nexmon_csi/files/4537534/config.log)

yujianyuanhaha, 2020-05-24:
Hello, I also met the issue that at final step, the tool `tcpdump` is not found on RT-AC86 Router, did you finally fixed that? @BonnieLi1 

however, this tool does exist on ubuntu machine at `/usr/sbin/tcpdump`, unfortunately, we are supposed to execute this tool [Usage: step 6](https://github.com/seemoo-lab/nexmon_csi) on RT-AC86 Router.

Thanks for @jlinktu answer, I did follow your instruction and the compile pass, but when I ssh to the  RT-AC86, i still did not found `tcpdump` installed.

Best Regards

tweigel-dev, 2020-12-03:
@zeroby0 I just retried the cross-compiling of nexon if this works currently and it is.
But the crosscompilation of tcpdump fails at the crosscomplie of LibPCAP with the same error discribed at #123.
I did everything within sudo su and start at step one at the bcm4366c0 tutorial of nexmoncsi and continue with the extention for tcp dump discribed here: #123 

`gcc --version -->  7.5.0`

If I configure the LibPCAP without the crosscompilation environments everything works perfectly but obviously with the output is x86  

I'am running out of ideas. I Will try to cross compile it at an arm based environment like pi3.

zeroby0, 2020-12-03:
I don't know why that's happening. What does the log file say? There should be a log folder with log files. Has g++ been installed and `source setup_env.sh` been run?

Here is the tcpdump executable I compiled. You can extract it and scp to `/jffs/` and it should run.

[tcpdump.zip](https://github.com/seemoo-lab/nexmon_csi/files/5635574/tcpdump.zip)

tweigel-dev, 2020-12-03:
Nice it works. Thanks!
It would be great if we can deliver the nexutil and tcpdump as bin for the asus at this repo, isn't it?

zeroby0, 2020-12-03:
Great idea! Those and dhd.ko. Would save a great deal of time for people.

SuzyLalaland, 2025-10-30:
> Also make sure the interface you want to use is up:
> 
> ```
> wl -i <interface> up
> wl -i <interface> radio on
> ifconfig <interface> up
> ```

why wl not found appears