seemoo-lab/nexmon_csi #35: Problem capturing CSI of 80MHz WiFi

Junye97, 2020-01-30:
Hi there, 

It seems that there is a problem with capturing the CSI of 80MHz WiFi. I am using Raspberry Pi 4B to capture the CSI, and my router is configured to be on channel 64 with 80MHz bandwidth (802.11ac mode). 

It seems that this problem only appears with 80MHz BW, if I changethe BW to either 20 or 40 MHz (keeping channel 64) it works fine. I have tried to include/exclude MAC filtering and there is no other filters configured. By the way I am running the latest Nexmon CSI with MAC filtering issue fixed. Please advise. 

Cheers

jlinktu, 2020-01-30:
The firmware of the Raspberry Pi 4B does not support chanspec 64/80 by default.
To enable channel 64 with 80MHz bandwidth add
```CH80MHZ_CHSPEC(58, WL_CHANSPEC_CTL_SB_UU),```
to _additional_valid_chanspecs_ in _regulations.c_:
https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/src/regulations.c#L195
If you are unsure if a chanspec is supported use `nexutil -k` after configuration to check if the correct channel was set.

Junye97, 2020-01-31:
Hi @jlinktu, thank you very much for your help. Just saying it seems to be a typo there it should be `CH80MHZ_CHSPEC(58, WL_CHANSPEC_CTL_SB_UU)`. 

Also, out of curiosity, what determines if a chanspec is supported? Does it vary by region (I'm in Australia).

matthiasseemoo, 2020-01-31:
Generally using our nexmon patch you can set any chanspec that the hardware
supports even 80mhz in the 2.4 GHz band which is not even part of the
standard. You need to check your local regulations on what channels you are
allowed to send with which bandwidth. If you want more details about it,
read our WiSec Paper on reactive jamming.

jlinktu, 2020-02-01:
@Junye97
`CH80MHZ_CHSPEC(`**`58`**`, WL_CHANSPEC_CTL_SB_UU),` is not a typo.
Channel specification 64/80 actually means channel number 58 for 80MHz (from 5250 to 5330 MHz, bonding the four 20MHz channels 52, 56, 60, and 64) using the 20MHz channel 64 as control channel.
_...CTL_SB_UU_ meaning control sideband upper upper, in this case channel 64, upper lower would be 60, lower upper 56 and lower lower 52.

Junye97, 2020-02-04:
Hi @jlinktu, Thank you very much for your clarification. @matthiasseemoo Thank you and I will have a look at the paper.

closed 2020-02-04 by Junye97

yujianyuanhaha, 2020-06-11:
do we need to modify `nexmon_csi/src/regulations.c` on ASUS AC86U router when aim at channel 157/80MHz ?

matthiasseemoo, 2020-06-11:
You should define it, as you cannot set any channel other than 20 MHz control channels by default.