seemoo-lab/nexmon_csi #36: 256 Subcarriers with 80MHz BW 802.11ac WiFi

Junye97, 2020-02-04:
Hi there, 

I have tried collecting CSI of a 80MHz 802.11ac WiFi, however, the output seems to be CSI of 256 subcarriers (read using the csiread.m MATAB script). 802.11ac should only have 128 subcarriers with 80MHz BW configuration. It seems that the number of subcarriers in the output is determined by: 

https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/utils/matlab/csireader.m#L19

with BW=80 the NFFT would be 256, and BW of 40 would give NFFT of 128, which do not match the  802.11ac/n standard. 

Cheers

jlinktu, 2020-02-04:
Hi,
you're wrong. The fft size is 64 for 20MHz, 128 for 40MHz, and 256 for 80MHz.
In 802.11-2016 see "21.3.7.2 Subcarrier indices in use" for VHT and "19.3.7 Mathematical description of signals" for non-HT and HT.

Junye97, 2020-02-05:
Oops.. thank you for the reply. 

Cheers

closed 2020-02-05 by Junye97