seemoo-lab/nexmon_csi #37: How to read csi on Raspberry Pi B4?

coldcoldmilk, 2020-02-04:
Hi,i have captured the DUP packets,is there a program to reading csi from a Raspberry Pi B4？What if not?

Junye97, 2020-02-05:
Hi there,

There is a Matlab script provided: ./src/matlab/csiread.m

Otherwise you can use apps like wireshark to inspect the raw pcap output. 

 Cheers

coldcoldmilk, 2020-02-08:
Thank you for your help.
However, I would like to be able to do everything in the Raspberry Pi, including reading csi information.But,i can't install matlab on the Raspberry Pi.
I'm very upset.
I'm trying to read ./src/matlab/csiread.m, and try to rewrite it in python.

coldcoldmilk, 2020-02-08:
> Hi there,
> 
> There is a Matlab script provided: ./src/matlab/csiread.m
> 
> Otherwise you can use apps like wireshark to inspect the raw pcap output.
> 
> Cheers

Thank you for your help.
However, I would like to be able to do everything in the Raspberry Pi, including reading csi information.But,i can't install matlab on the Raspberry Pi.
I'm very upset.
I'm trying to read ./src/matlab/csiread.m, and try to rewrite it in python.

Junye97, 2020-02-08:
MATLAB is a very demanding program... If you are looking for realtime CSI decoding that might be a lot to ask for a Pi, the next best thing in my opinion is to open the port 5500 and have a remote device that does it.

mzakharo, 2020-02-08:
Rpi is perfectly capable for realtime CSI decoding, if you keep your sounding rate low, like 10-100HZ. Just make sure you use the compiled C decoder.

coldcoldmilk, 2020-02-13:
> MATLAB is a very demanding program... If you are looking for realtime CSI decoding that might be a lot to ask for a Pi, the next best thing in my opinion is to open the port 5500 and have a remote device that does it.

Thank you very much for your answer.

closed 2020-02-13 by coldcoldmilk

zeroby0, 2020-02-13:
Also, people seem to be facing issues with Pi 4. #41 #39
I'm using a pi 3B+ and everything seems to be going good, but I've not collected 40K packets like these people did. If you do get into problems, check using a 3B+ too.

mzakharo, 2020-02-14:
This is not a 3+ vs 4 issue. Crash occurs if you get amsdu and duplicate frames. Not everyone operates in wifi environments that capture those.