seemoo-lab/nexmon_csi #38: Generating frames for CSI capture

princeulrich2, 2020-02-09:
Hello,
Thanks very much for this amazing tool. I am able to capture CSI on a raspberry pi 3B+ using a TP-LINK AC750 router on a 20MHz bandwidth.

However I wait for several minutes before getting some packets and sometimes just a few are captured.
I understand that I need to generate the Wi-Fi frames from the router so I tried a few things like downloading large files, streaming videos in multiple browsers and pinging the  the router's IP at a regular rate using the command below but the capture still stalls:

`sudo ping -i 0.05 ROUTER_IP`

My filter parameters are as follows:
`makecsiparams -e 1 -c $CHANNEL/$BANDWIDTH -C 1 -N 1 -m $ROUTER_MAC_ADDRESS`

Could you please suggest a way to generate these frames? Maybe someone is able to capture a stream of frames? Thanks in advance

zeroby0, 2020-02-09:
Packets are probably going through the 5 GHz band, and you're listening on the 2.4 GHz.
Set channel and bandwidth to your routers 5 GHz band and check?

princeulrich2, 2020-02-09:
Thanks for your reply, I appreciate. I changed to channel 36 on the 5 GHz band and even disabled the 2.4 GHz band but I still get the same issue. I am able to capture packets without generating the frames and when it stalls, all the efforts to generate frames as mentioned above do not yield anything.

Junye97, 2020-02-10:
Hi @princeulrich2, 

Maybe check if the 5 GHz channel is supported by default? You can refer to my earlier issue #35 to see how to add channel supported. If I were you, I would start with 2.4 GHz and see if I can get any packet at all first. 

Cheers

princeulrich2, 2020-02-11:
Hi @Junye97 , thanks for your reply.
I have already checked and I am able to capture packets on both the 2.4 GHz(20 MHz bandwidth) and the 5 GHz band(both 20 MHz and 80 MHz bandwidths). The issue is that the capture stops after some packets or it sometimes takes long to capture packets. I would like to control the generation of frames and if possible the rate at which they are generated.

zeroby0, 2020-02-11:
I'll try to summarise everything you said to see if I understood correctly. Correct me if I have not.

1. You're able to capture packets on both 20 and 80 MHz on channel 36.
2. You're not using any packet filter like #33 
3. When you're capturing, with tcpdump I assume, packet collection stops after sometime.
4. You're using the latest version of Nexmon, nexmon patch, and the kernel available.
5. Similar behavior is observed without the macID filter, and with any other router.

If everything is true, please answer these questions:
1. How many packets were you able to capture before it stops?
2. If you restart your pi and reload the driver, does everything start working again?

If your answer to 2 is 'yes', probably there is some kind of buffer that is full and cannot process more packets. Try using tcpdump with -w to write to a file.

If there is a bug in the firmware, one of the maintainers will probably fix it. I suggest you re-flash the SD card and setup from scratch and check once again. This has fixed problems a similar problem of mine in the past. Here is my build script.

Junye97, 2020-02-11:
I also experience the buffering issue but not as bad as @princeulrich2 's case. For me each single capture can last quite long (12 hours+ in one case), the issue rises when I try to restart capturing (reconfigure the mackcsiparams and run nexutil) multiple times without rebooting the Pi.

princeulrich2, 2020-02-12:
Hi @zeroby0 , thanks very much for your detailed input, I appreciate. I am using the kernel version is 4.19.75-v7+.

1. You're able to capture packets on both 20 and 80 MHz on channel 36.
- yes indeed. on channel 11 for the 20MHz and channel 36 for the 80MHz bandwidths.
2. You're not using any packet filter like #33
- yes 
3. When you're capturing, with tcpdump I assume, packet collection stops after sometime.
- yes
4. You're using the latest version of Nexmon, nexmon patch, and the kernel available.
- yes I reinstalled nexmon from scratch
5. Similar behavior is observed without the macID filter, and with any other router.
- yes I got a similar result with a linksys N600 router. Nothing is captured if I don't set the macID. 

Before the capture stopped the largest amount I got was about 46000 packets in about 13 minutes. Sometimes it stopped after capturing just 300 or 1430 packets so it varies. For the 5GHz band, if I restart the pi and reload the driver the capture always works and stops after some time, on average 10 - 15 minutes. 

For the 2.4 GHz band, restarting and reloading the pi does not always make it work. It takes long to capture any packet and stops after capturing a few, sometimes just about 5 packets would be captured. I am not sure why this is the case but I will use the 5GHz band for now. 

Thanks for sharing your script, it is very similar to the script which I use. 
I already write the output of tcpdump to a file.

princeulrich2, 2020-02-12:
Thanks @Junye97 . Are you also using a raspberrypi 3B+?
Indeed I tried to flood the WiFi with frames by using the ping command I mentioned earlier but it does not work for me. Does this work for you? or are you using a different command?

mzakharo, 2020-02-12:
@Junye97 , I have the same behaviour on RPI 4B. Check out my investigation in #41.  If you see a trap in `forensics` or your nexmon -k returns strange values once capture stops, you may be experiencing similar problems.

princeulrich2, 2020-02-12:
@mzakharo yes I also experienced this a couple of days ago

Junye97, 2020-02-13:
@princeulrich2 I am running a R Pi 4B with 4GB of ram. Maybe it is the ram that's limiting the capturing? But for me the ram usage does not go very high so I'm not sure about that. 

For me I usually generate ping packets using 

```sh
sudo ping 192.168.1.1 -i 0.02 
```

which basically generates ping packets to the router at 50Hz. I will have a look into @mzakharo 's forensics later and see if I have the same issue.

princeulrich2, 2020-02-13:
I see, I am trying to generate packets in a similar way, at 20Hz and varying the frequency does not yield a different outcome... The ram usage is also not very high so I do not think that it can be the issue.

Junye97, 2020-02-14:
@princeulrich2 I noticed that you mentioned if there is no MAC address filter configured you will not get any CSI. Please refer to my previous issue #21 . There was a bug in the code so if you can pull the Nexmon CSI repo again you should be able to get CSI without MAC address.

princeulrich2, 2020-02-15:
yes indeed you are right @Junye97, thank you. I pulled the repo again after re-flashing the sd card and I am able to capture CSI without the MAC address filter. This time the capture lasts longer, I was able to capture for more than 3 hours. For now the result seems to be the same whether I use the ping command to generate the packets or not so I cannot conclude that I am able to generate packets. I will try further because I really need to be able to generate packets (and at a specific rate if possible).

Junye97, 2020-02-18:
@princeulrich2 for me ping with -i option does the job more or less, unless you are asking for some insanely high frequency or accuracy. One thing you should consider is definitely the packet header filter, which can help you get rid of things like wifi control packets

princeulrich2, 2020-02-18:
@Junye97 I would also expect that it works for me, given that I'm trying to generate at a low frequency of 20Hz... Currently I do not use the packet header filter because I would like to get all the available packets but I will try your suggestion

zeroby0, 2020-02-18:
@princeulrich2 I assumed you're using a third device to ping the router and generate data. Are you doing that, or is your raspi connected to wifi when you collect data, and you use it to ping?

princeulrich2, 2020-02-18:
@zeroby0 Yes I am using a third device(laptop) to ping the router. wpa_supplicant is killed on the raspberry pi. Sometimes I also try to use the raspberry pi to ping the router but it is connected via a network cable, not WiFi.