seemoo-lab/nexmon_csi #39: Nexutil sets chanspec failed

zlrpuck, 2020-02-11:
I tried use command nexutil -k157/80 to change the channel spec to Channel 157 and 80MHz bandwidth. 
And then I use nexutil -k to check if I had set successfully. The output it not 157/80 but 
chanspec: 0x6863, 85/160.

Anyone has the same issue here?
Thanks

zeroby0, 2020-02-11:
85 isn't even a defined channel, so I'm amazed that that happened. Trying to set that manually gives me `invalid chanspec`. However, I'm able to set 157/80 on my raspi 3b+. If you rebooted your pi, did you load the modified firmware again?

zlin-ir, 2020-02-11:
After I reboot my RPi4, I install the firmware patches again.
The command nexutil -k157/80 works at beginning and I can capture CSI data by tcpdump.
However, after few minutes (not sure about the exact number), the issue above shows up again and tcpdump stops working.
[image]

zlrpuck, 2020-02-11:
@zeroby0 
After I reboot my RPi4, I install the firmware patches again.
The command nexutil -k157/80 works at beginning and I can capture CSI data by tcpdump.
However, after few minutes (not sure about the exact number), the issue above shows up again and tcpdump stops working.
[image]

mzakharo, 2020-02-11:
This maybe a duplicate of  #41.   I just ran `nexutil -k` after my firmware traps, and it also takes longer than usual  I also get the 85/160 nonsense.

zlin-ir, 2020-02-13:
@zeroby0 First of all, thanks for your detailed scripts posted at #38 .
Reboot sometimes solve the problem (not always, sometimes I have to reboot several times).

zeroby0, 2020-02-13:
I don't think this will solve it, but how about unloading and reloading the driver from the kernel? 

While compiling nexmon_csi, if you do `make backup-firmware` before `make install-firmware`, it should copy the stock firmware for backup. Replace nexmon_csi with stock, and then stock with nexmon_csi.

But this probably is an issue that only occurs on pi4. As @mzakharo said, looks like 41 is related. 

@jlinktu, will you please take a look at this when you're free?

mzakharo, 2020-02-14:
I doubt 3+ would help. The issue was first reported under pi 3+ in parent nexmon repo

jlinktu, 2020-02-14:
As mentioned earlier 0x6863, 85/160 is indeed not a valid chanspec. 0x63 0x68 is in fact the hex ascii of "ch", beginning of the string "chanspec" in the ioctl buffer. This return value most likely indicates that the chip is down as it should be overwritten by the current chanspec. Hence @mzakharo might be right that this is connected to #41, but it could also be something different. Please have a look at the output of
``` cat /sys/kernel/debug/ieee80211/phyX/forensics``` (where X is your current phy)
to see if there's a crash dump.

zlin-ir, 2020-02-16:
Thanks @jlinktu 
Please kindly take a look at the output of my forensics attached below.
Frankly, I don't quite understand the content. 
[receivedinfo.txt](https://github.com/seemoo-lab/nexmon_csi/files/4209432/receivedinfo.txt)

zlin-ir, 2020-02-20:
cat /sys/kernel/debug/ieee80211/phy1/forensics
`dongle trap info: type 0x4 @ epc 0x0022a9d2
  cpsr 0x0000019f spsr 0x000001bf sp 0x0025feb0
  lr   0x001a12b3 pc   0x0022a9d2 offset 0x25fe58
  r0   0x00000000 r1   0x002266fc r2 0x0022abfc r3 0x00000011
  r4   0xbd98f199 r5   0x00000000 r6 0x0025a654 r7 0x00000000

000547.142 unexpected new csi, clearing old
000547.143 unexpected new csi, clearing old
000547.143 unexpected new csi, clearing old
000547.144 unexpected new csi, clearing old
000547.144 unexpected new csi, clearing old
000547.145 unexpected new csi, clearing old
000547.147 unexpected new csi, clearing old
000547.147 unexpected new csi, clearing old
000547.148 unexpected new csi, clearing old
000547.148 unexpected new csi, clearing old
000547.149 unexpected new csi, clearing old
000547.149 unexpected new csi, clearing old
000547.149 unexpected new csi, clearing old
000547.149 unexpected new csi, clearing old
000547.159 unexpected new csi, clearing old
000547.162 unexpected new csi, clearing old
000547.163 unexpected new csi, clearing old
000547.164 unexpected new csi, clearing old
000547.164 unexpected new csi, clearing old
000547.164 unexpected new csi, clearing old
000547.164 unexpected new csi, clearing old
000547.168 unexpected new csi, clearing old
000547.170 unexpected new csi, clearing old
000547.170 unexpected new csi, clearing old
000547.171 unexpected new csi, clearing old
000547.171 unexpected new csi, clearing old
000547.171 unexpected new csi, clearing old
000547.171 unexpected new csi, clearing old
000547.174 unexpected new csi, clearing old
000547.174 unexpected new csi, clearing old
000547.191 
FWID 01-e1db26e2
flags 1
000547.191 
TRAP 4(25fe58): pc 22a9d2, lr 1a12b3, sp 25feb0, cpsr 19f, spsr 1bf
000547.192   dfsr 801, dfar bd98f1cf
000547.192   r0 0, r1 2266fc, r2 22abfc, r3 11, r4 bd98f199, r5 0, r6 25a654
000547.192   r7 0, r8 23d638, r9 25a63c, r10 22abfc, r11 4013, r12 0
000547.192 
   sp+0 00000000 0023d73c 0000000e 00000013
000547.192   sp+10 00004000 0022aa73 0025ff5c 0025a654

000547.192 sp+44 001c1c47
000547.192 sp+7c 001c87f1
000547.192 sp+a4 001a6939
000547.192 sp+b4 001a6975
000547.192 sp+c4 0019a4d9
000547.192 sp+cc 0019aba9
000547.192 sp+dc 00199c47
000547.192 sp+e8 000001df
000547.192 sp+128 00008221
000547.192 sp+138 0019ca0f
`

zlin-ir, 2020-02-20:
cat /sys/kernel/debug/ieee80211/phy1/forensics
`dongle trap info: type 0x4 @ epc 0x0022a9d2
  cpsr 0x0000019f spsr 0x000001bf sp 0x0025feb0
  lr   0x001a12b3 pc   0x0022a9d2 offset 0x25fe58
  r0   0x00000000 r1   0x002266fc r2 0x0022abfc r3 0x00000011
  r4   0xbd98f199 r5   0x00000000 r6 0x0025a654 r7 0x00000000`

`000547.142 unexpected new csi, clearing old
000547.143 unexpected new csi, clearing old
000547.143 unexpected new csi, clearing old
000547.144 unexpected new csi, clearing old
000547.144 unexpected new csi, clearing old
000547.145 unexpected new csi, clearing old
000547.147 unexpected new csi, clearing old
000547.147 unexpected new csi, clearing old
000547.148 unexpected new csi, clearing old
000547.148 unexpected new csi, clearing old
000547.149 unexpected new csi, clearing old
000547.149 unexpected new csi, clearing old
000547.149 unexpected new csi, clearing old
000547.149 unexpected new csi, clearing old
000547.159 unexpected new csi, clearing old
000547.162 unexpected new csi, clearing old
000547.163 unexpected new csi, clearing old
000547.164 unexpected new csi, clearing old
000547.164 unexpected new csi, clearing old
000547.164 unexpected new csi, clearing old
000547.164 unexpected new csi, clearing old
000547.168 unexpected new csi, clearing old
000547.170 unexpected new csi, clearing old
000547.170 unexpected new csi, clearing old
000547.171 unexpected new csi, clearing old
000547.171 unexpected new csi, clearing old
000547.171 unexpected new csi, clearing old
000547.171 unexpected new csi, clearing old
000547.174 unexpected new csi, clearing old
000547.174 unexpected new csi, clearing old
000547.191 
FWID 01-e1db26e2
flags 1
000547.191 
TRAP 4(25fe58): pc 22a9d2, lr 1a12b3, sp 25feb0, cpsr 19f, spsr 1bf
000547.192   dfsr 801, dfar bd98f1cf
000547.192   r0 0, r1 2266fc, r2 22abfc, r3 11, r4 bd98f199, r5 0, r6 25a654
000547.192   r7 0, r8 23d638, r9 25a63c, r10 22abfc, r11 4013, r12 0
000547.192 
   sp+0 00000000 0023d73c 0000000e 00000013
000547.192   sp+10 00004000 0022aa73 0025ff5c 0025a654`

`000547.192 sp+44 001c1c47
000547.192 sp+7c 001c87f1
000547.192 sp+a4 001a6939
000547.192 sp+b4 001a6975
000547.192 sp+c4 0019a4d9
000547.192 sp+cc 0019aba9
000547.192 sp+dc 00199c47
000547.192 sp+e8 000001df
000547.192 sp+128 00008221
000547.192 sp+138 0019ca0f`