seemoo-lab/nexmon_csi #41: firmware trap on RPI4

mzakharo, 2020-02-11:
Firmware traps occurs on Raspberry Pi 4B.

Steps to reproduce:
1. start capture on 157/80 with mac/frame filtering disabled. 
2. connect a client (phone) to a router (ASUS AC-68U), configured to broadcast @ 157/80.  Use the client to navigate to [fast.com](fast.com)
3. firmware traps within seconds.

I replaced sdio.c with [this](https://hastebin.com/raw/gacemubala). 

dmesg:
```
[ 4426.093512] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
[ 4426.161399] brcmfmac: CONSOLE: 4(25fe58): pc 22a9d2, lr 1a12b3, sp 25feb0, cpsr 19f, spsr 1bf
[ 4426.161408] brcmfmac: CONSOLE: 000066.620   dfsr 801, dfar bd98f1cf
[ 4426.161415] brcmfmac: CONSOLE: 000066.620   r0 0, r1 2266fc, r2 22abfc, r3 11, r4 bd98f199, r5 0, r6 25a654
[ 4426.161422] brcmfmac: CONSOLE: 000066.620   r7 0, r8 20bfec, r9 25a63c, r10 22abfc, r11 4013, r12 0
[ 4426.161427] brcmfmac: CONSOLE: 000066.620 
[ 4426.161433] brcmfmac: CONSOLE:    sp+0 00000000 0020c0f0 0000000e 00000013
[ 4426.161439] brcmfmac: CONSOLE: 000066.620   sp+10 00004000 0022aa73 0025ff5c 0025a654
[ 4426.161445] brcmfmac: CONSOLE: 
[ 4426.161450] brcmfmac: CONSOLE: 000066.620 sp+44 001c1c47
[ 4426.161456] brcmfmac: CONSOLE: 000066.620 sp+78 00004013
[ 4426.161462] brcmfmac: CONSOLE: 000066.620 sp+7c 001c87f1
[ 4426.161467] brcmfmac: CONSOLE: 000066.620 sp+a4 001a6939
[ 4426.161473] brcmfmac: CONSOLE: 000066.620 sp+b4 001a6975
[ 4426.161479] brcmfmac: CONSOLE: 000066.620 sp+c4 0019a4d9
[ 4426.161484] brcmfmac: CONSOLE: 000066.620 sp+cc 0019aba9
[ 4426.161490] brcmfmac: CONSOLE: 000066.620 sp+d0 00004013
[ 4426.161496] brcmfmac: CONSOLE: 000066.620 sp+dc 00199c47
[ 4426.161501] brcmfmac: CONSOLE: 000066.620 sp+e8 000001df
[ 4426.161507] brcmfmac: CONSOLE: 000066.620 sp+11c 00004013
[ 4426.161513] brcmfmac: CONSOLE: 000066.620 sp+128 00008221
[ 4426.161519] brcmfmac: CONSOLE: 000066.620 sp+138 0019ca0f
```
cat /sys/kernel/debug/ieee80211/phy1/forensics
```

dongle trap info: type 0x4 @ epc 0x0022a9d2
  cpsr 0x0000019f spsr 0x000001bf sp 0x0025feb0
  lr   0x001a12b3 pc   0x0022a9d2 offset 0x25fe58
  r0   0x00000000 r1   0x002266fc r2 0x0022abfc r3 0x00000011
  r4   0xbd98f199 r5   0x00000000 r6 0x0025a654 r7 0x00000000
_rx: bad frame length (1799)
000062.113 wl0: dma_rx: bad frame length (1799)
000062.508 wl0: dma_rx: bad frame length (1799)
000062.511 wl0: dma_rx: bad frame length (1799)
000063.004 wl0: dma_rx: bad frame length (1799)
000063.007 wl0: dma_rx: bad frame length (1799)
000063.109 wl0: dma_rx: bad frame length (1799)
000063.112 wl0: dma_rx: bad frame length (1799)
000063.113 wl0: dma_rx: bad frame length (1799)
000063.506 wl0: dma_rx: bad frame length (1799)
000063.508 wl0: dma_rx: bad frame length (1799)
000064.007 wl0: dma_rx: bad frame length (1799)
000064.008 wl0: dma_rx: bad frame length (1799)
000064.123 wl0: dma_rx: bad frame length (1799)
000064.126 wl0: dma_rx: bad frame length (1799)
000064.506 wl0: dma_rx: bad frame length (1799)
000064.509 wl0: dma_rx: bad frame length (1799)
000065.006 wl0: dma_rx: bad frame length (1799)
000065.008 wl0: dma_rx: bad frame length (1799)
000065.508 wl0: dma_rx: bad frame length (1799)
000065.511 wl0: dma_rx: bad frame length (1799)
000066.004 wl0: dma_rx: bad frame length (1799)
000066.009 wl0: dma_rx: bad frame length (1799)
000066.503 wl0: dma_rx: bad frame length (1799)
000066.506 wl0: dma_rx: bad frame length (1799)
000066.620 PKTGET failed
000066.620 dngl_sendpkt: dropped pkt
000066.620 
FWID 01-e1db26e2
flags 1
000066.620 
TRAP 4(25fe58): pc 22a9d2, lr 1a12b3, sp 25feb0, cpsr 19f, spsr 1bf
000066.620   dfsr 801, dfar bd98f1cf
000066.620   r0 0, r1 2266fc, r2 22abfc, r3 11, r4 bd98f199, r5 0, r6 25a654
000066.620   r7 0, r8 20bfec, r9 25a63c, r10 22abfc, r11 4013, r12 0
000066.620 
   sp+0 00000000 0020c0f0 0000000e 00000013
000066.620   sp+10 00004000 0022aa73 0025ff5c 0025a654

000066.620 sp+44 001c1c47
000066.620 sp+78 00004013
000066.620 sp+7c 001c87f1
000066.620 sp+a4 001a6939
000066.620 sp+b4 001a6975
000066.620 sp+c4 0019a4d9
000066.620 sp+cc 0019aba9
000066.620 sp+d0 00004013
000066.620 sp+dc 00199c47
000066.620 sp+e8 000001df
000066.620 sp+11c 00004013
000066.620 sp+128 00008221
000066.620 sp+138 0019ca0f
```

mzakharo, 2020-02-11:
Ok, I was able to apply some fixes and 'fast.com' trick no longer triggers an immediate crash.

Applying changes [here](https://github.com/seemoo-lab/nexmon_csi/compare/master...mzakharo:master) and adding '-d 50' to makecsiparams, I am able to run much longer,

However, CSI traffic still eventually stops, now without a trap.
Firmware seems to be stopped - judging by `/sys/kernel/debug/ieee80211/phy1/counters` are not changing.
Output of  `/sys/kernel/debug/ieee80211/phy1/forensics`  yields:

`wlc_recvamsdu: wrong A-MSDU deagg sequence, cur_state=0`

Maybe this needs proper solution, as per https://github.com/seemoo-lab/nexmon/issues/280

`

jlinktu, 2020-02-14:
Does this occur with commit 32378c1feb9ef7d01e069c42035650f1f75b8bb2 and earlier too (prior to the latest ucode changes)?

mzakharo, 2020-02-20:
I copied monitor.c from nexmon/ folder, and  it seems to work without issues if I stick with 157/80 spec -> has been running for more than 7 days now. So it seems like  AMSDU https://github.com/seemoo-lab/nexmon/issues/280#issuecomment-516731866 workaround does the trick.

```
//Workaround to skip AMSDU frames. 
__attribute__((at(0x1B6B02, "", CHIP_VER_BCM43455c0,FW_VER_7_45_189)))
BPatch(wlc_monitor_amsdu_patch, 0x1B6B1E);
```

closed 2020-02-20 by mzakharo