seemoo-lab/nexmon_csi #42: Can I use an external adapter to stay connected to network?

zeroby0, 2020-02-11:
So, we need to turn off wpa_supplicant before collecting data. If I have an external USB WiFi dongle, is there a way to have that connected to WiFi and use the built-in chip to collect CSI data?

I'm trying to get it working, and will add my findings here as a reference for future users. For my use-case, it's okay if the dongle has to connect to an open network of WPS.

zeroby0, 2020-02-12:
The simplest method I've found so far is to use an Android phone and do USB tethering. Works out of the box. Port forwarding apps can be used to ssh into the pi.

Junye97, 2020-02-13:
If you can have android tethering via USB cable, there is nothing stopping you using the ethernet port on the Pi. If you have to use WiFi connection, I think a USB WiFi dongle will just be fine, albeit you need configure the connection without wpa_supplicant.

zeroby0, 2020-02-13:
@Junye97 Ethernet works, but I'd need to sacrifice mobility. I'm trying to localise a Pi indoors with CSI in realtime, so a link to the network to relay collected data will be nice. Also, since I'm still in development, I need continuous shell access to monitor, and to change settings.

I'm investigating WiFi dongles, but some of them need recompiling the kernel and such. The ones that work out of the box are designed at the time of Pi 2, and have only 802.11 n. This issue thread is to document my attempts which could be useful to future users. As a part of that, your suggestions are really welcome 😄

zeroby0, 2020-02-13:
There are some projects that use dkms and seem to be easier to use
- https://github.com/aircrack-ng/rtl8812au
- https://github.com/gnab/rtl8812au
- https://github.com/abperiasamy/rtl8812AU_8821AU_linux
- https://github.com/diederikdehaas/rtl8812AU
- https://github.com/svpcom/rtl8812au

I've installed aircrack-ng/rtl8812au, and was able to scan networks, but didn't get to connecting it to a WiFi. Looks like WPA cannot be used at the moment because we're removing wpa_supplicant. WPS might still be possible, but it's not very secure, and might not be worth setting up. 

SSH tunnels with ssk-keys are secure over public networks, so the lack of WPA is probably okay. Also,[ ip traffic can be tunnelled over ssh](https://sshuttle.readthedocs.io/en/stable/) if you need network.

jlinktu, 2020-02-14:
You can also try to keep wpa_supplicant running but use e.g. wpa_cli to disable things like autoscan for the interface connected to the wifi chip used as extractor. It might be enough to stop it from interfering. Have not tested it but might work.

zeroby0, 2020-02-14:
@jlinktu  gonna test it out first thing in the morning!

zeroby0, 2020-02-22:
Apparently, you can [disable wpa_supplicant on one interface](/etc/dhcpcd.conf) by adding this to `/etc/dhcpcd.conf`
```
interface wlan0
        nohook wpa_supplicant
```

I've just gotten the wifi adapter to work with this driver: https://github.com/aircrack-ng/rtl8812au. The installation was failing because my kernel updated.

When the dongle is plugged in, it's recognised as phy0, and is assosciated with wlan0. But that could probably be reversed with plugging the dongle after boot completes. I'll add the procedure if everything works out.

zeroby0, 2020-05-28:
To conclude this thread, I have not gotten an external adapter to work. There are adapters that seem to work with Pi out of the box, and it may not be worthwhile to try to get other adapters to work. I think Adafruit has one that would work.

I connected the raspi to an old Android with USB tethering to give it internet access. The Raspi is behind a NAT created by the phone, so to access it, you should use a port forwarding app and forward Raspi IP's port 22 to some ip on your phone. It's yet another hassle to find the IP of your Raspi as it's randomised on reconnect. I connected to the Pi via ethernet to find it's USB tethering IP and setup port forwarding. It was not worth it.

If I have to do it again, I'd get a [very long Ethernet cable, and a Power over Ethernet kit](https://www.amazon.in/hz/wishlist/ls/1PA5Q279W8GD5?&sort=default) and save myself days worth of time. The Ethernet cable is likely longer than the range of a WiFi adapter anyway. If you're going this way, make sure you choose your PoE equipment carefully, because the Pi's Network stack doesn't work properly when inadequately powered.

@Junye97 Oh how I wish I'd just heeded your advice..

closed 2020-05-28 by zeroby0

YuhaoWangPolyU, 2020-07-08:
> @Junye97 Ethernet works, but I'd need to sacrifice mobility. I'm trying to localise a Pi indoors with CSI in realtime, so a link to the network to relay collected data will be nice. Also, since I'm [...]

Dear,

Can we use this patch to collect the pcap data in real-time? If so, how can we transmit the data?

zeroby0, 2020-07-08:
I use an Ethernet cable to send data from Pi to my computer. CSI data is indeed collected in realtime into a pcap, but I transfer it to my laptop after the collection is complete.

YuhaoWangPolyU, 2020-07-08:
dear

Are there any method that we can send the package to laptop in real time?

已從 Outlook Mobile<https://aka.ms/blhgte> 傳送

[https://www.polyu.edu.hk/emaildisclaimer/PolyU_Email_Signature.jpg]

Disclaimer:

This message (including any attachments) contains confidential information intended for a specific individual and purpose. If you are not the intended recipient, you should delete this message and notify the sender and The Hong Kong Polytechnic University (the University) immediately. Any disclosure, copying, or distribution of this message, or the taking of any action based on it, is strictly prohibited and may be unlawful.

The University specifically denies any responsibility for the accuracy or quality of information obtained through University E-mail Facilities. Any views and opinions expressed are only those of the author(s) and do not necessarily represent those of the University and the University accepts no liability whatsoever for any losses or damages incurred or caused to any party as a result of the use of such information.

zeroby0, 2020-07-09:
I think you can do that with tcpdump. I didn't do it, so I don't know the details, but google "wireshark remote capture", that would be a good starting point. If it works, please post the solution here.