seemoo-lab/nexmon_csi #43: What does the X-axis represent in the graph?

coldcoldmilk, 2020-02-13:
Hi,thank you for your help.
What does the X-axis represent?Why do subcarriers have negative Numbers?

zeroby0, 2020-02-13:
I was inclined to say 'subcarriers', since the range of the plot is same as the bandwidth I selected for capture, but I'm not sure anymore.

Sure, the bandwidth is 80, but that's made of 4 20MHz channels. So I should see a drop in amplitude at every 20 MHz on the plot, but I see a drop at 30 on my data.

[image]

Also, the graph is half empty, but that is probably because the devices need not always use the complete bandwidth. Probably the associated paper: https://dl.acm.org/doi/10.1145/3349623.3355477 has info. Please post your findings.

Junye97, 2020-02-14:
Generally the x axis refers to the subcarrier indices. @zeroby0 They are not empty, but rather the channel with significant amplitude you were looking at are actually *unused* subcarriers. You can refer to ieee 802.11-2016 standards and find _*21.3.7.2 Subcarrier indices in use*_ section with tells you the exact subcarriers used. After removing the unused subcarriers you will see the CSI amplitude plots much nicer.

princeulrich2, 2020-02-15:
Yes indeed it refers to the subcarrier indices. The null subcarrier is in the middle and the indices of the others are relative to that, hence the negative values. The picture below shows how they are organized in a 20 MHz channel. 
[image]
It is from https://www.researchgate.net/publication/330496708_Innovative_Methods_for_Fair_Coexistence_between_LTE_and_Wi-Fi_in_Unlicensed_Spectrum. 
@zeroby0 just a small remark, it seems that the data used in the plot above was collected from a 40MHz channel instead of 80MHz, because I see 128 subcarriers which corresponds to a 40MHz channel, instead of the 256 subcarriers in a 80MHz channel. Indeed if you remove the unused subcarriers you will get a better result

closed 2020-03-01 by coldcoldmilk