seemoo-lab/nexmon_csi #44: I/Q swapped on bcrm43455c0 (RPi3/4)?

mzakharo, 2020-02-13:
Hello,

The comment  in https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/src/csi_extractor.c#L219

Implies that the code below is placing imaginary into the lower 16 bit. 
 https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/src/csi_extractor.c#L224-L225 

complex numbers  are normally packed  with real in the lower address.

jlinktu, 2020-02-14:
Hi,
extracted CSI, 32bit per subcarrier, hold the quadrature (imaginary) part in the lower 14bit and the in-phase (real) part in the 14bit above. We convert them to 16bit each for easier processing, keeping them in order. This I-Q-order is quite common, but if you prefer to switch real and imaginary part in your project you can easily do that without breaking anything.

mzakharo, 2020-02-14:
Then isnt the sample matlab code incorrect then?
  Also,  switching downstream is doable, but not 'efficient'. Below code is vectorized conversion of 16 bit array to numpy:

`H = np.frombuffer(buffer, dtype='<h').astype(np.float32).view(np.complex64)`

Data being in Q/I order prevents usage of this vectorized API.

jlinktu, 2020-02-15:
True. The order is wrong in the example for bcm4339 and bcm43455c0, and it makes more sense to have I/Q order in the UPD payload. For now you can just swap lines https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/src/csi_extractor.c#L222 and https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/src/csi_extractor.c#L224.
Will probably push it here soon too.

Junye97, 2020-02-27:
Hi @jlinktu, does that imply in order to get the correct phase information, we will need to swap the real and imaginary component in MATLAB for processing?