seemoo-lab/nexmon_csi #48: Whether bcm43455c0 can connect to wifi after running 'ifconfig wlan0 up'?

whusober, 2020-02-27:
Hi there,
I am trying to use nexmon to get the CSI information on raspberry Pi 4b. Having done all the options mentioned by [nexmon_csi](https://github.com/seemoo-lab/nexmon_csi) , I ran tcpdump but got nothing. I suspect that I have done something wrong with the commond `ifconfig wlan0 up` whose front is `pkill wpa_supplicant` , because the computer just cannot reconnect to wifi after I let wlan0 up. Could you please tell me if this is normal?

Junye97, 2020-02-28:
Hi there, 

The wifi adaptor is running in monitor mode, thus it cannot connect to any wifi.

whusober, 2020-03-08:
> Hi there,
> 
> The wifi adaptor is running in monitor mode, thus it cannot connect to any wifi.

Thanks! With your help, I solves this problem.

closed 2020-03-08 by whusober