seemoo-lab/nexmon_csi #49: How to control the delivery rate of UDP packets？

coldcoldmilk, 2020-03-01:
Hello, the paper you gave me last time has solved my question,thanks a lot. @Junye97 @princeulrich2

Now I want to use the extracted CSI data to do some research.

I found that I was using the code `tcpdump -i wlan0 DST port 5500`

The frequency of UDP packet acquisition is not controlled

So,how to control the delivery rate of UDP packets？for example, I want to get 10 UDP packets in one second.

Junye97, 2020-03-02:
Hi there, 

The easiest way is to generate ping packets. You can run 

```sh
sudo ping <some_ip> -i <ping_interval>
```

If you need exactly 10 packets per second, you will need some filtering since there will be control and acknowledge packets transmitted as well.

coldcoldmilk, 2020-03-02:
> Hi there,
> 
> The easiest way is to generate ping packets. You can run
> 
> ```shell
> sudo ping <some_ip> -i <ping_interval>
> ```
> 
> If you need exactly 10 packets per second, you will need some filtering since there will be control and acknowledge packets transmitted as well.

Hi,
How should I choose the IP address?
Should I use my raspberry PI to select my router's IP address?

dejavo4e, 2021-04-12:
> 
> 
> Hi there,
> 
> The easiest way is to generate ping packets. You can run
> 
> ```shell
> sudo ping <some_ip> -i <ping_interval>
> ```
> 
> If you need exactly 10 packets per second, you will need some filtering since there will be control and acknowledge packets transmitted as well.

hi; what is the default value for the packets per second? Is is dependent on ping command? For 20MHz bandwidth, we have a 1000x64 matrix. Each row is CSI packet ,i guess.
columns are sub-carriers. So where is time?
thanks in advance

parisafm, 2021-04-12:
> hi; what is the default value for the packets per second? Is is dependent on ping command? For 20MHz bandwidth, we have a 1000x64 matrix. Each row is CSI packet ,i guess.
> columns are sub-carriers. [...]

hi; hope these explanations are useful for you: 
Each column is subcarrier. Depending on PHY standard, u can have different numbers of subcarriers. ( as seen in the picture).
[image]

For ping commands check this webpage: 
https://www.ionos.com/digitalguide/server/tools/ping-command/

About ur last question, i may have same issue. 
Looking forward to hearing from Junye97 or others.