seemoo-lab/nexmon_csi #52: What is wrong with my graph?

KvN-Sparrow, 2020-03-09:
Greetings!

First of i'd like to thank everyone that contributed to this project.

I have configured a RPi B3+ as an AP operating on ch.36 using the 802.11ac 5Ghz wireless mode and I am then using a RPi B4 as the CSI extractor. (no frame filtering)

I successively manage to capture some packets with tcpdump yet once plotting the measurements within Matlab the resulted graph varies significantly from the one created using the example.pcap file.

example.pcap
[image]
vs mine
[image]

Does this suggest an error in setup/configuration or are there any subsequent steps required in order to clean the data from unused sub-carriers?

Any sort of feedback is appreciated:)

Junye97, 2020-03-16:
Hi there, 

Technically there is nothing wrong with your plot. In short, the unused subcarriers (the yellow ones on the left and subcarrier 0) are in your plot so they are messing up the amplitudes. Please refer to #43 for more details. 

Cheers

jlinktu, 2020-03-16:
The provided example.pcap was captured on a Nexus 6P containing data in a floating point format.
Data captured on a Raspberry Pi is in a simpler integer format.
Your plot of the example.pcap does not represent the contained CSI as you've probably processed the data wrong.

The plot of your data looks fine. As @Junye97 suggested you should consider removing or zeroing out the null/guard carriers.