seemoo-lab/nexmon_csi #54: Does this plot shows the real CSI that we caught?

whusober, 2020-03-22:
Hi there,
Our team have run the nexmon_csi on rasp-Pi 4B and successfully draw plots of the data. But no matter how we walk betwwen the AP and computer, the data plot doesn't change a lot.  As you can see in the plot, the 0 subcarrier reaches a high level, but we get to know from the wiki pedia that the 0 subcarrier should stay 0. So I am confused whether we got the correct CSI data or not?
[image]

RoshanCM1998, 2020-03-22:
Hey Whusober,
Roshan Here, Can you provide me your email ?
I am also starting to use this tool and getting error and you setup tool successfully so i can get some help from you.

jlinktu, 2020-03-22:
@whusober 
Remove or set to zero the null/dc carrier and the guard carriers as they contain arbitrary values. Changes in the CSI will be better visible afterwards.

You can find the valid data&pilot subcarrier indices i.a. here: [802.11ac: A Survival Guide by Matthew S. Gast, Table 2-1](https://www.oreilly.com/library/view/80211ac-a-survival/9781449357702/ch02.html#table-channel-description).

For 20MHz HT/VHT that would mean setting the subcarriers with indices -32 to -29, 0, and 29 to 31 to zero.

whusober, 2020-03-23:
> Hey Whusober,
> Roshan Here, Can you provide me your email ?
> I am also starting to use this tool and getting error and you setup tool successfully so i can get some help from you.

Of crouse my friend , contact me by whusober@gmail.com

whusober, 2020-03-23:
> @whusober
> Remove or set to zero the null/dc carrier and the guard carriers as they contain arbitrary values. Changes in the CSI will be better visible afterwards.
>
> You can find the valid data&pilot [...]

Thanks! I will try it.

closed 2020-03-29 by whusober

Z-Yannn, 2022-12-07:
> Thanks! I will try it.

Hi my friend, did you solve the issue finally? I am also stuck in the same problem and I am not sure how can I set the null/dc carrier. Could you please give me some tips?

sarielkk, 2023-08-29:
> > > @whusober
> > > 删除零/直流载波和保护载波或将其设置为零，因为它们包含任意值。之后 CSI 的变化将会更加明显。
> > > 您可以在此处找到有效的数据和导频子载波索引 ia：[802.11ac：Matthew S. Gast 的生存指南，表 2-1](https://www.oreilly.com/library/view/80211ac-a-survival/9781449357702/ch02.html#table-channel-description)。
> > > 对于 20MHz HT/VHT，这意味着将索引为 -32 至 -29、0 和 29 至 31 的子载波设置为零。
> > 
> > 
> > 谢谢！我会尝试。
> 
> 你好朋友，请问你的问题最后解决了吗？我也遇到了同样的问题，我不知道如何设置零/直流载波。您能给我一些建议吗？

Hello, did you solve the problem?

Z-Yannn, 2023-08-29:
> Hello, did you solve the problem?

Hi, I did not set the dc/null subcarriers to 0 during the collection period. Instead, I stored all the information in the data frame and simply skipped the useless subcarriers in the data preprocessing part.

xaxm007, 2024-11-13:
> Hi, I did not set the dc/null subcarriers to 0 during the collection period. Instead, I stored all the information in the data frame and simply skipped the useless subcarriers in the data [...]

My csv file of the CSI data indexes the subcarrier from 0 to 63, would you mind helping me how i can remove the null/dc and pilot subcarrier based on those indexes?