seemoo-lab/nexmon_csi #55: Problem to collect csi

RoshanCM1998, 2020-03-22:
I am using Raspberry Pi 4 B(bc43455c0).
"Getting Started" setup is done well but I am getting problem to collect csi data

makecsiparams -c 157/80 -C 1 -N 1 -m 00:11:22:33:44:55,aa:bb:aa:bb:aa:bb -b 0x88 
in this command for -c it is written that 157 is channel and 80 is bandwidth, 
but here what I have to put as channel ? Is it on which router is currently working or on which I want to collect csi ?
And if I want to collect collect csi data on more then one channel then how to do I mean In other csi collection tool like "Atheros csi tool" they provide csi data of 54 channels output at 20MHz bandwidth and 124 channels output at 40MHz bandwidth at ones then how to do this kind of setup ?

What exactly -C (coremask) and -N (nssmask) represent and what I provide value to this ?

In tcpdump it print all packets on terminal, so to use this packets what I have to do I mean, I need to store in any .txt file or they automatically stored in any file ?

A-T-Kristensen, 2020-03-22:
Hi Roshan, the Atheros tool cannot collect from 54 channels at the same time, they collect from 56 sub-carriers for a 20 MHz channel. See [link](https://github.com/xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src/wiki/Getting-started-with-Atheros-CSI-tool) at the part  "Atheros-CSI-Tool reports the CSI value for each subcarrier, i.e., 56 subcarriers for 20MHz channel and 114 subcarriers for a 40MHz channel."

You can see a list of number of sub-carriers for different PHY standards here https://www.oreilly.com/library/view/80211ac-a-survival/9781449357702/ch02.html

For the remainder of your questions, I cannot give a qualified answer, although I think -N is used to control spatial streams.

jlinktu, 2020-03-22:
@RoshanCM1998 
The wifi chip of the raspberry pi will tune to the channel given by parameter _-c_ _\<channel\>\/\<bandwidth\>_ before starting to collect CSI. Thus it is the channel you want to collect CSI on.

You are confusing channels with subcarriers.
The _channel_ defines the center frequency the chip shall operate on and the bandwidth defines how much spectrum around the center frequency shall be used. This spectrum around the center frequency is divided into a specific amount of _subcarriers_ (frequencies that are orthogonal to each other).  For 20MHz those are 64, for 40MHz 128, and for 80MHz 256 subcarriers. Not all of them are data/pilot subcarriers which you are interested in, thus the actual amount is slightly less, see #54. 
For a list of WLAN channels you can have a look at https://en.wikipedia.org/wiki/List_of_WLAN_channels. 

The _coremask_ given with _-C_ tells the chip of which receive front ends/antennas it shall extract CSI. As the raspberry pi has only you are supposed to use 1 here.
The _nssmask_ given with _-N_ tells the chip of what spatial streams it shall extract CSI.
For more information on how spatial streams and multiple antennas are used in 802.11 you might want to have a look at [802.11-2016](https://standards.ieee.org/standard/802_11-2016.html) or resources describing it.

You can use tcpdump's parameter _-w_ to write packets to a _pcap_ file. See https://www.tcpdump.org/manpages/tcpdump.1.html for more information.

RoshanCM1998, 2020-03-22:
Thank you for replay A-T-Kristensen and jlinktu,
Yes, it was my mistake to write channel instead of subcarriers.
I understand all of this quit well thanks.

RoshanCM1998, 2020-03-22:
nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==

jlinktu, please can you tell me about this command ?

jlinktu, 2020-03-22:
_nexutil_ makes an ioctl call 500 to the wifi chip holding the data given by _-v_, where _-b_ tells nexutil that the value is in base64 and _-l34_ specifies the ioctl buffer length to be 34 bytes.
It is used to tell the firmware how to configure the CSI extractor, you can see how this is handled here:
https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/src/ioctl.c#L125
The value is a filled out version of the following struct in base64 and can be produced using _makecsiparams_:
https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/utils/makecsiparams/makecsiparams.c#L44

RoshanCM1998, 2020-03-22:
when i try to run tcpdump -i wlan0 with "dst port 5500" filter then no packets were received.
I try to set makecsiparams as my router configuration like -c 157/40 -C 1 -N 1 -m as router mac address and didn't set -b.
I try to run tcpdump many time out of that 1-2 times some packets were received, all other time not a single packet was received.
So please can you explain explain me reason of  this. And please guide me so that it became easy to record csi packets when I want to record( I mean from a separate router I can send a packet when I want by ping or any thing else) so that it can be possible to record csi data and I can generate a dataset for ML model.

jlinktu, 2020-03-22:
There are many possible reasons.
As already discussed in several other issues make sure the chip really tunes to the specified channel (you might need to add the chanspec to the firmware first), check if the mac address really matches, make sure there is traffic.

RoshanCM1998, 2020-03-24:
In makecsiparams -b param is to filter packets base on it's starting hex like 0x88 so here I want to generate and sent a unique packet when ever I want and record csi of that particular packet.
please guide me to do this or some thing like this.

WangShanyue, 2020-04-13:
Have you solved it ? I met the same problem with you using Raspberry Pi 3+.

WangShanyue, 2020-04-14:
I realized the issues I met were discussed already!😂