seemoo-lab/nexmon_csi #56: Asus router installation timeout

bachejames, 2020-03-22:
Hi there,

Thanks so much for putting this together - really cool!

I am however stuck when trying to install on the Asus Router. During execution of step 7 (`make install-firmware REMOTEADDR=<address of your rt-ac86u>`), I get the following:

This line runs for about 130 seconds: `COPYING TO ROUTER dhd.ko -> /jffs/dhd.ko` before I then get this:
`ssh: connect to host <address of your rt-ac86u> port 22: Connection timed out 
lost connection`

Any ideas why please? It is perhaps a router setting I can change to increase time before timeout?

Thanks so much :)

jlinktu, 2020-03-22:
Did you substitute _\<address of your rt-ac86u\>_ with the IP address of your router?
If not, you are supposed to do so.

bachejames, 2020-03-22:
Resolved - had to turn ssh on

closed 2020-03-22 by bachejames