seemoo-lab/nexmon_csi #58: 40MHz chanspec on 2.4GHz on bcm4358 (nexus 6P)

mzakharo, 2020-04-03:
Hello,

I am trying to set channel spec for 40MHz on channel 1u  on 2.4GHz. My channel specs keep getting rejected by the firmware.

As per ```CH40MHZ_CHSPEC(1, WL_CHANSPEC_CTL_SB_U);``` , channel spec for this should be ```0x1901```

However, when I upload csi params, firmware log shows this:

`wl0:` invalid chanspec 0x1901`

 `wl0: wlc_iovar_op: chanspec BCME -20 (Bad Channel)`

I noticed that in https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/src/regulations.c#L238  does not have a hook for bcm4358. Is this something that can be added to resolve this?

Thanks,
`

jlinktu, 2020-04-05:
Hi,

unfortunately there are no further flashpatches available for the bcm4358.
But one solution might be to remove this flashpatch:
https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/src/regulations.c#L81
And instead hook the `wlc_valid_chanspec_db` function:
```
int
wlc_valid_chanspec_db_hook(void *wlc_cm, unsigned short chanspec)
{
    int valid = wlc_valid_chanspec_ext(wlc_cm, chanspec, 1); 
    int i;
    if (!valid)
        for (i = 0; i < sizeof(additional_valid_chanspecs)/sizeof(additional_valid_chanspecs[0]); i++)
            valid |= additional_valid_chanspecs[i] == chanspec;
    return valid;
}
__attribute__((at(0x5132C, "flashpatch", CHIP_VER_BCM4358, FW_VER_7_112_300_14)))
BPatch(wlc_valid_chanspec_db_hook, wlc_valid_chanspec_db_hook);
```

You'll also have to add the address of `wlc_valid_chanspec_ext` for the bcm4358 to `nexmon/patches/common/wrapper.c`. The function is already in there.
```
AT(CHIP_VER_BCM4358, FW_VER_7_112_300_14, 0x4FE6C)
```

mzakharo, 2020-04-06:
HI @jlinktu , thank you for your prompt response. These tips definitely helped get past the error in the log, however, the chip does not seem to be happy operating at 40MHz -> it does not seem to capture most packets. I get something through occasionally, but at much lower rate than if I select 20MHz. Not sure whats the limiting factor is here.