seemoo-lab/nexmon_csi #59: Separating out signals from different antennas

bachejames, 2020-04-05:
Hey! Thanks again for the awesome tool :)

I am looking to estimate the angle of arrival of signals by separately analysing the signal received by each of the Asus RT-AC86U's three antennas. Here is my setup:
Receiver: `Asus RT-AC86U`
coremask: `-C 3`
nssmask: `-N 1`
Code to identify from which antenna the signal is arriving at: [from here](https://github.com/seemoo-lab/nexmon_csi/issues/51#issuecomment-602220951)

This setup is giving me a core value alternating between 0 & 1, and a constant rxss value of 0. I was however expecting a core value alternating between values: 0, 1 and 2 representing the signal received from each of the three antennas in turn.

I'm guessing that it's because I don't properly understand the `coremask` and `nssmask` settings correctly. My current understanding is:
The `coremask` value represents the number of antennas that should record data, however this hypothesis is inconsistent with me receiving values of only 0 or 1 rather than 0, 1 or 2.
The `nssmask` value represents the number of spatial streams to capture - and am guessing this means that this value should match the number of antennas the transmitter has (as long as it supports that number of streams).

I'm also a little confused by the fact that in the `makecsiparams -h` readme, it seems to indicate that these two values should be inputted in binary format, however all the examples I've seen on GitHub just use the decimal `1`.

Many thanks for your help!

mzakharo, 2020-04-05:
well, the word 'mask' -> means that each bit in a value represents an antenna. 0x1 -> 1 . 0x3 -> 2 bits -> 2 antennae, 0x7 -> 3 bits -> 3 antenna. Hope this helps!  You can even do something like 0x4 -> antenna 3 only, or something exotic like 0x5, which is is antenna 1 and 3.

Here is more info:
https://en.wikipedia.org/wiki/Mask_(computing)

jlinktu, 2020-04-05:
Hi,

`coremask` and `nssmask` do not only set the number of cores and streams to use but also specify of what cores and streams csi shall be extracted.
Values for both `-C` and `-N` can be any number between 1 and 15, thus all 4 bit combinations excluding 0, where the bits set to 1 indicate the indices of the cores or streams to extract from.

If you want to extract of the stream with index 0 of cores with indices 0, 1 and 2 you need to set `-C 7` and `-N 1`. Decimal 7 equals to binary `0111`, thus setting bits for core 3 to `0`, core 2 to `1`, core 1 to `1` and core 0 to `1`.

Btw. the rt-ac86u should actually have four antennas, one being hidden inside the case.

bachejames, 2020-04-06:
Very many thanks @jlinktu and @mzakharo, really appreciate the super helpful, very s[eedy replies.

closed 2020-04-06 by bachejames

mikeWangJC, 2020-04-30:
@bachejames
hi, buddy
I also meet the similar problem.
I cannot collect CSI data with three antennas of Asus RT-AC86U simultaneously, via

 ./makecsiparams -c 149/80 -C 7 -N 1 MAc-address -b 0x80

 Do you have any suggestions? 
 Thank u so much.

bachejames, 2020-05-02:
Hey @mikeWangJC I haven't actually tried using all three antennas, but from my limited knowledge I can only suggest experimenting with turning on and off different antennas and seeing which provide the readout you are hoping for. 

For example, perhaps instead of `0111` ie. `-C 7`, the right combination may be `1110` ie. `-C 14`.

mikeWangJC, 2020-05-05:
@bachejames
thank you so much for ue suggestion. 
BTW, I want to form a transceiver pair to collect CSI data. Do you know how to control the bandwidth and frequency of the signal?

bachejames, 2020-05-16:
I don't I'm afraid, sorry @mikeWangJC

matthiasseemoo, 2020-05-17:
https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_firmware/blob/master/src/ioctl_5xx.c#L489 <https://github.com/seemoo-lab/wisec2017_nexmon_jammer_demo_firmware/blob/master/src/ioctl_5xx.c#L489>

yujianyuanhaha, 2020-06-09:
> Hi,
>
> `coremask` and `nssmask` do not only set the number of cores and streams to use but also specify of what cores and streams csi shall be extracted.
> Values for both `-C` and `-N` can be any [...]

May I know the concept of `core`, what does it stand for?

The `nssmask` is more clear to me,  for 2 router, each have 4 anttennas, there could be at most 4*4 = 16 spatial stream.