seemoo-lab/nexmon_csi #61: Which MAC adress should follow the makecsiparams -m?

whusober, 2020-04-15:
Hey guys,

We are using nexmon_csi to track CSI information between a server and a client. Unfortunately, we got two totally different results when we used the srever's MAC and client's MAC as -m of makecsiparams.

In our understanding, the MAC should be the server's MAC, but the true results are below (We run a script on the server that sending the UDP packets in a particular size and speed):

When using the client's MAC:
[image]

When using the server's MAC:
[image]

As you can see, it seems that the first one is our expection, but it is contrary to our understanding on makecsiparams.

Could you please tell us which MAC adress should follow the -m of makecsiparams?

jlinktu, 2020-04-15:
Hi,
the addresses given with `-m` are compared against the second address in the MAC header of incoming frames. CSI are extracted if they match.

whusober, 2020-04-23:
> Hi,
> the addresses given with `-m` are compared against the second address in the MAC header of incoming frames. CSI are extracted if they match.

Could you please give us your email address😂We have made a report on our problem, and we want to send this report to you, hope you can give us some pointers🌹

jlinktu, 2020-04-23:
https://github.com/seemoo-lab/nexmon_csi/blob/b52fca3abc18715d6d12692e531164b5d62a78fd/README.md#L167