seemoo-lab/nexmon_csi #62: 0 packets captured

289536718, 2020-04-20:
Hi

I have successfully installed the firmware and script on the RI 4b, and can also successfully start the monitoring mode. But when I started monitoring mode, I couldn't receive the package. I have changed the channel several times, or this result, what should I do?
I hope someone can help me, thank you very much.

Thank you.

```
root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/makecsiparams# tcpdump -i mon0 dst port 1 -w /home/pi/xxxx.pcap

tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
0 packets captured
0 packets received by filter
0 packets dropped by kernel
```

mikeWangJC, 2020-04-21:
i hav the same problem.
It seems that RI 4b does not communicate with the router, so no packet is captured.
When I opened the tool, my RI 4b could not initiate a data request, which may be the reason why the router does not send RI 4b a packet

mikeWangJC, 2020-04-21:
Is it possible to solve this problem by forming a transceiver pair, for example,  RI 4b is the transmitter and another ASUS router is the receiver.
In intel 5300 CSI tool, I completed the data collection by forming a transceiver pair.

289536718, 2020-04-21:
@mikeWangJC 
Thank you for your reply. I haven't found a way to solve this problem yet. I have installed the system according to the suggestions in the discussion area, but I still can't grasp the package.

jlinktu, 2020-04-21:
Hi,
first of all, when you are in monitor mode you are not supposed to be connected to an access point, what is not impossible but I recommend using the CSI extractor as passive monitor only. When you are in monitor mode you'll see all frames on the channel you are tuned to, not only those destined to you, thus it does not matter if you are connected to the access point as long as there are other devices creating traffic. If you want to control the traffic, simplest is to use a device that supports frame injection (e.g. by using nexmon) to transmit specific frames, or have two devices (e.g. an access point and a second raspberry) communicate over their wireless interfaces (e.g. by using `ping`, `iperf`, ... ).

The default `nexmon` patch for the Raspberry Pi 3B+/4B supports multiple interfaces, meaning you can have `wlan0` as normal interface and `mon0` as monitor interface.
That is not really needed for the `nexmon_csi` patch and also not implemented. Just stick with `wlan0` (e.g. when calling tcpdump). I admit the README might be misleading here as an interface `mon0` is created but all it does is enable monitor mode in the firmware. Instead of creating the `mon0` interface just using `nexutil -Iwlan0 -m1` should also be fine.

mikeWangJC, 2020-04-22:
@jlinktu 
man, thank you so musch for ur recommendation. i've collected CSI successfully after i removed :
-m mac addr.
So, I mean, does this command  prevents the raspberry from receiving the packet from the AP with this address

289536718, 2020-04-22:
@jlinktu 
Thank you for your reply. Like mikeWangjc, I removed the MAC address when generating a Base64 encoded parameter string. Later, I also successfully caught the bag and collected CSI.

jlinktu, 2020-04-22:
Regarding _-m_ see https://github.com/seemoo-lab/nexmon_csi/issues/61#issuecomment-614183155, it does not prevent receiving any frames but controls for what frames CSI is extracted,  make sure to use the correct mac addresses.
Your second problem is the same as https://github.com/seemoo-lab/nexmon_csi/issues/10#issuecomment-550404402, just with 80MHz instead of 40MHz.

289536718, 2020-04-23:
@jlinktu 
Mr.Link, Thank you for your answer.

I've found the right MAC address, on 36 / 80, and got the package. But I still have a problem
1. After adding the 0x88 filter, I can't collect any packets, and my router is being used. With 0x80, I can collect some packets. Without any filter, I can collect more packets, but after using the tcpdump command several times, it will stop, and the channel will jump to a strange parameter, so I have to restart RI 4. However, the use of filters will not cause this problem.

I modified CSI extractor. C with reference to #46 , but I will still repeat this situation. So I hope you can explain:

1. Whether the CSI of bencon and QoS exist in different channels, which leads to channel confusion when I do not use any filter to extract packets.

2. What should I do if I don't want channel jump?

3. If I want to extract the CSI of QoS, what should I do?

jlinktu, 2020-04-23:
1. Use any device in monitor mode to observe if the frames you are expecting can be found on the channel. (You can use the Raspberry Pi with CSI extraction patch for this, just omit the ioctl 500 and set the channel with `nexutil -k<channel/bandwidth>` instead.)
If you see the expected frames use the first byte of these frames as argument to the filter option. 0x80 are beacon frames and are most probably sent over 20MHz on the control channel. 0x88 is not the only data frame type.
If you can not see the expected frames you might not be in range, the frames you are looking for do not exist at all, or you are observing the wrong channel.
2. In a normal setup there shouldn't be any channel hopping. To make sure your CSI extractor stays on its channel kill wpa_supplicant or suppress autoscans.
3. If you want a specific frame type to be sent use frame injection.

289536718, 2020-04-24:
@jlinktu 
Thank you for your reply. I have some problems to solve. Next, I plan to use ASUS ac-86u and RI4 to form a transmit-receive. Thank you again.

closed 2020-04-24 by 289536718