seemoo-lab/nexmon_csi #63: How are the CSI packets timestamped?

Junye97, 2020-04-23:
Hi there,

To my understanding, each captured CSI frame contains its own epoch timestamp. My question is, is each timestamp fetched directly from the system (e.g. from Linux), or only one timestamp is fetched while the rest are calculated? How does this play out when I update the system time during CSI collection? I’m running Nexmon CSI on a Raspberry Pi 4B. Thank you very much.

Cheers

mzakharo, 2020-04-23:
I believe the timestamps are part of https://github.com/torvalds/linux/blob/master/Documentation/networking/timestamping.txt

Namely, ```SOF_TIMESTAMPING_RX_SOFTWARE```

Junye97, 2020-04-23:
Thank you for your reply. So this basically means the timestamps are generated Individually by requesting the timestamp from the OS? But I couldn’t find any code in Nexmon repo with 

``
SOF_TIMESTAMPING_RX_SOFTWARE
``

mzakharo, 2020-04-24:
Thats because the source code for SOF_TIMESTAMPING is entirely inside Linux kernel source. all nexmon does is generate a UDP packet, and pass it over to linux networking stack. rest you can follow from linux sources.

Junye97, 2020-04-24:
Thank you for your help. I'm closing the issue.

closed 2020-04-24 by Junye97