seemoo-lab/nexmon_csi #64: BCM4354 for Nvidia TX2

xuhao1, 2020-04-24:
Dear Developers,
 I am new to this area, I wonder is that possible to use this software for BCM 4354 chipset on Nvidia TX2, which is one of the most widely used onboard computer fo movable robots?
If not, how to modify the code to make it running, I am happy to contribute code after I finish this.

jlinktu, 2020-04-25:
Hi,

the current patches only support bcm4339, bcm43455c0, bcm4358 and bcm4365/6.

The bcm4354 is not added to the basic nexmon project yet, that would be a first step to bring the CSI extractor to it. That means finding several memory regions and pointers, function addresses and struct field offsets. After that you'll have to patch the ucode that runs on the d11 core to extract CSI and find some more patch specific addresses. If you are running L4T you'd also have to patch the brcmfmac driver to get nexmon working.

If you don't want to or cannot do this by yourself contact me via mail at jlink@seemoo.tu-darmstadt.de.

xuhao1, 2020-04-28:
> Hi,
> 
> the current patches only support bcm4339, bcm43455c0, bcm4358 and bcm4365/6.
> 
> The bcm4354 is not added to the basic nexmon project yet, that would be a first step to bring the CSI extractor to it. That means finding several memory regions and pointers, function addresses and struct field offsets. After that you'll have to patch the ucode that runs on the d11 core to extract CSI and find some more patch specific addresses. If you are running L4T you'd also have to patch the brcmfmac driver to get nexmon working.
> 
> If you don't want to or cannot do this by yourself contact me via mail at [jlink@seemoo.tu-darmstadt.de](mailto:jlink@seemoo.tu-darmstadt.de).

Thanks for your reply, I will take a look at the basic nexmon project.