seemoo-lab/nexmon_csi #66: Asus RT-AC86U AP Mode

chaosmaster, 2020-04-27:
I am running the csi-patches on an Asus RT-AC86U.
As soon as the patches are loaded, I can't seem to connect to the SSID.
It seems to always be stalling when trying to obtain an IP address.
Without the patches connecting works fine.
CSI extraction does work and I can see the SSID.
Is this expected behavior or am I missing something?

jlinktu, 2020-04-28:
Expected. Devices in monitor mode are not supposed to be connected to an AP.

chaosmaster, 2020-04-28:
OK, thanks for the reply.
So AC86U can only collect CSI while in monitor mode?
On RaSP4 i was able to be in AP mode and still collect CSI.

jlinktu, 2020-04-28:
It should still be possible to serve as AP and be in monitor mode.
I guess you have to reinit all the AP related stuff after reloading the dhd.ko.

chaosmaster, 2020-04-28:
I have it set up, that it will load dhd.ko on boot (replacing it in /lib/modules)
And I can see the SSID, so it is in AP mode.
But for some reason I am unable to connect to it.