seemoo-lab/nexmon_csi #67: __nex_driver_io: error //  Channel cannot be switched

289536718, 2020-04-28:
@jlinktu 
I'm sorry to bother you again, sir, but I have a problem that needs to be solved.Similar to #17  and #34 . After entering the router and using the nexutil command, I encountered the error of `"nex driver" IO`, but I couldn't solve it. I tried the following:
1. Before installing the nexmon firmware, I have installed the latest version of Merlin firmware on my AC86U, and after installing nexmon, I have installed tcpdump into the router.

2. I have tried the solutions you proposed in the previous problems (including downloading nexmon again, starting the eth5&&eth6), but still failed.

I have the following problems:
After entering the JFFS directory of the router, attempting to enter the following command will result in the following error:
1. If you use nexuti to view the channel, an error will be reported and the channel cannot be switched
```
admin@RT-AC86U-9808:/jffs# ./nexutil -k
__nex_driver_io: error
chanspec: 0x6863, 85/160
admin@RT-AC86U-9808:/jffs# ./nexutil -k36/80
__nex_driver_io: error
```

2. Use nexutil to match the parameters. Whether you add or remove - S500 is still normal, but after you remove S500, dmesg will report an error
```
admin@RT-AC86U-9808:/jffs# ./nexutil -Ieth6 -s500 -b -l34 -vKu*****AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
admin@RT-AC86U-9808:/jffs# ./nexutil -Ieth6 -b -l34 -vKu******AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
admin@RT-AC86U-9808:/jffs# dmesg
.
.
.
CONSOLE: 028449.710 wl1: invalid chanspec 0xe02a
dhd_prot_ioctl: status ret value is -17
dhd_prot_ioctl: status ret value is -17

```
I don't know what to do now. I hope you can help me.

Thank you very much.

jlinktu, 2020-04-28:
1. Include the interface name in the nexutil call, e.g. `nexutil -Ieth6 -k`.
2. Do not remove _-s500_.

289536718, 2020-04-29:
@jlinktu 
Thank you, sir
According to your instructions, I have successfully collected CSI.
But I have the following problems to solve:

1. I want to estimate the angle of arrival of signals by separately analysing the signal received by each of the Asus RT-AC86U's three antennas. According to the method given by you in #59 , I have made the following settings:
`makecsiparams -c 149/80 -C 7 -N 1 -m MAC_address -b 0x80`
After collecting the package and running it in MATLAB's 
 `readpcap.m`
In  readpcap.m
```
% data link type
In obj. Global_header. Network = freed (obj. FID, 1, '* uint32');
```
network shows 1

So, I hope you can give me the answer:

1. Whether I set the parameters correctly. If not, what to do if you want to use 3 antennas at the same time?

2. How to modify the code of MATLAB to get the correct image in plot? I can't see it if I add three lines of code directly in #51  according to you.
```
payloadbytes = typecast(payload,'uint8');
core = bitand(payloadbytes(56),3);
rxss = bitand(bitshift(payloadbytes(56),-3),3);
```
Thank you very much

mikeWangJC, 2020-04-30:
@jlinktu @289536718 
I think I meet the same  problem.
I'm not sure if i hav collected the CSI by using three antennas simultaneously via the command

makecsiparams -c 149/80 -C 7 -N 1 -m MAC_address -b 0x80

Also, I hav no idea how to process (or check) if the data is collected sucessfully via three antennas,  by using the provided MATLAB code.
Hope you guys can give me some suggestion. 
Thank you so much.

mikeWangJC, 2020-04-30:
hi guys， 
 I have successfully collected data with three antennas，simultaneously.
The problem is the data collected by the third antenna is always zero!!!
I have checked that there is no problem with this antenna.
 This is unreasonable， does anyone know why ??

Junye97, 2020-04-30:
@mikeWangJC i would check if the third spatial stream exists at all. I used the router to collect a single spatial stream CSI and only one antenna returns non-zero values. Alternatively, you can inspect the pack files using wireshark.

mikeWangJC, 2020-05-06:
@Junye97 
Thank u for ur suggestion.  I will carefully check whether the third antenna exists。
BTW, I'm trying to form a transceiver pair by using the an router and raspberry pi.
In the process, I failed to control the signal bandwidth and frequency. @jlinktu 
Have u ever met such problem? do u have any suggestion?

289536718, 2020-05-06:
@mikeWangJC 
谢谢您的回复，您的问题我也想要解决，现在我开了新的链接#69

yujianyuanhaha, 2020-05-29:
@289536718 Hello, I also met the same problem of `__nex__driver__io: error` 

Does install MERLIN is a must?  @mikeWangJC @289536718 @jlinktu  I am working on official firmware _3.0.0.4.384_32799-gfe72567_, everything look fine till I fail at execute `nexutil`

289536718, 2020-06-01:
@yujianyuanhaha
Sure. Install the merlin can open the jffs in your ac86u.

jlinktu, 2020-06-04:
no, merlin is not required.

yujianyuanhaha, 2020-06-06:
> no, merlin is not required.
> regarding `__nex__driver__io: error` make sure the modified firmware is loaded and to use the correct interface name with `nexutil`.

yes, i later notice the error come from I have to `make install-firmware ...` after the router reboots

closed 2020-06-07 by 289536718

B6G-K1LLER, 2024-08-27:
I got monitor mode working and airdump but while using reaver it shows that __nex__driver__io: error for some reason.

I have samsung s6 edge rooted - stock nougat rom - custom aldeon kernel 3.10.61

Wifi : bcm4358a3