seemoo-lab/nexmon_csi #69: Enter the injection mode.

289536718, 2020-05-06:
@jlinktu 
Sir, I'm trying to combine two nexmon devices into a transceiver pair, because I want to control the bandwidth, frequency and transmission rate of the transmitted signal, but I don't know how to enter the injection mode. Can you help me?
Thanks.

mikeWangJC, 2020-05-06:
Greetings!
I meet the same problem as you do. 
I find that the data I collect  usually only covers 20MHz bandwidth, but I want to collect 80MHz data.
So I tried to use ASUS nd Raspberry Pi to form a transceiver pair.
But I fail to collect the data I want. Do u have any suggestion for injecting packet?
Thanks in advance!

289536718, 2020-05-06:
I just found that nexmon and nexmon_csi are two different firmware. I always thought that the latter is a supplement to the former. Maybe we can go into the injection mode through nexmon.

closed 2020-05-07 by 289536718